speeduino/speeduino #6: Discussion: Missing tooth is offset ATDC in speeduino but BTDC in megasquirt

mumme74, 2015-02-19:
The setting Trigger Angle is defined as After TDC on cyl 1 in speeduino. 
On most engines i have seen (mostly car engines) the missing tooth is defined as Before TDC on cyl 1.

See Delay teeth drawing in megamanual :
http://www.megamanual.com/ms2/wheel.htm

[image]

And more about the this here:
http://www.extraefi.co.uk/wheel_fitting.htm

[image]

Note the setting is BTDC not ATDC

noisymime, 2015-02-19:
Yes, this was intentional. To me, referencing the trigger based on BTDC always seemed a little backwards and I couldn't see any good reason why Megasquirt did it this way as opposed to ATDC. 

It's trivial to change to BTDC if you think it's preferable, but I'm not sure what the advantage is.

mumme74, 2015-02-20:
>Yes, this was intentional. To me, referencing the trigger based on BTDC always seemed a little backwards and I couldn't see any good reason why Megasquirt did it this way as opposed to ATDC.

Ok I find it quite the opposite, when a car engines flywheel has sync pulse 60deg before TDC it would seem strange to set the value to 300deg instead of just 60. But like you said, its no big issue.

>It's trivial to change to BTDC if you think it's preferable, but I'm not sure what the advantage is.

Well the advantage would be that it works the same way as other systems.
It will probably be easier to point future users to documentation pages like the ones above

noisymime, 2016-06-27:
Closing this as this is working per design. Understand that other systems may work differently, but Speeduino will continue to use ATDC

closed 2016-06-27 by noisymime