speeduino/speeduino #7: errors while compiling

TheJamm, 2015-04-06:
Hi im getting the error messages bellow while trying to upload/compile not sure what to do, a bit of help would be appreciated

corrections:146: error: 'PID' does not name a type
corrections.ino: In function 'byte correctionsAFRClosedLoop()':
corrections:189: error: 'egoPID' was not declared in this scope
display.ino: At global scope:
display:6: error: 'Adafruit_SSD1306' does not name a type
display.ino: In function 'void initialiseDisplay()':
display:13: error: 'display' was not declared in this scope
display:26: error: 'display' was not declared in this scope
display:26: error: 'SSD1306_SWITCHCAPVCC' was not declared in this scope
display:29: error: 'WHITE' was not declared in this scope
display.ino: In function 'void updateDisplay()':
display:38: error: 'display' was not declared in this scope

noisymime, 2015-04-06:
Did you copy the libraries folders across? Step one here: http://speeduino.com/wiki/index.php/Compiling_and_Installing_Firmware#Compiling_the_firmware

TheJamm, 2015-04-06:
Thanks Thats sorted it. now i just need to make a board and try it on an engine, what input sensors do you recomend?

noisymime, 2015-04-07:
Take a look through here for the sensors: http://speeduino.com/wiki/index.php/Hardware_requirements

The most important is probably the crank sensor and missing tooth trigger arrangement. After that you either need a MAP sensor or a TPS for load control. 

As far as boards go, the v0.3 is now available for purchase if you're interested: https://speeduino.com/shop/index.php?id_product=8&controller=product

closed 2015-04-07 by noisymime