speeduino/speeduino #51: Some ideas to improve the speed

tausday, 2016-10-18:
Hello,

I'm new at Speeduino and i have to admit that i don't understand really much, but i will try to help to make it better.

Now i will tell u, which ideas i have.

1. Increase the speed from the loop. The speed of the loop is slow because there is an overhead, if u write something like:

while (true)
    {
        // do stuff...
    }

The Loop will be in while and it will be much faster.

2.  Change from digitalWrite() to digitalWriteFast() thats an library which increase the speed by change the pin by manipulation.
digitalWrite() = 6280ns
digitalWriteFast() = 125ns
I have read this article about the accuracy of the ignition timing.. maybe its makes better or did u have an correction factor for the slow pinchange.
http://speeduino.com/forum/viewtopic.php?t=223

3. sensors.ino
There are some code what i not understand:
For all sensors(TPS,CLT,IAT,O2,Bat) u read it first time without doing something with the valu:
Example;

analogRead(pinTPS);
byte tempTPS = fastMap1023toX(analogRead(pinTPS), 0, 1023, 0, 255); //Get the current raw TPS ADC value and map it into a byte

Why is the code "analogRead(pinTPS);" in row 92?? I don't understand why u do that, because in the next row u read it again.

4. Maybe its make faster if the ADC Ports run in free so the code will not be stopped just be interrupted.

I hope u understand me.

Best regards 

tausday

VitorBoss, 2016-10-20:
If you look closer the code have a lot of analogRead() repeated for same input.

Thank you for digitalWriteFast() idea, I have tested it and is even more effective, it scores about 58 times faster. It's a wonder.

tausday, 2016-10-20:
thanks 4 response..
Victor can u tell me why there is an "standalone" analogRead? analogRead(pinTPS); i would be understand it and i can't sleep because that

noisymime, 2016-10-20:
With the 2 calls to analogRead, this is to ensure accuracy of the reading. When doing a number of calls to analogRead for different pins, you have to ensure that the internal sampling capacitor has enough to to full charge and hence get an accurate reading. 
analogRead itself does not run for a fixed amount of time, it has a loop that tracks the status of this ADC read and returns when ready. The 2nd call to analogRead runs many many times faster than the first.

Just generally around performance improvements, there is a gauge you can view in TunerStudio that shows the loops per second that the arduino is performing. **This is the primary indicator of performance in Speeduino**. Load up a running test system (ie has an RPM input from something like ardustim and is outputting fuel/ignition pulses) and see what that gauge reads. Make a change somewhere in the code, upload it and see what difference it makes on that gauge.

VitorBoss, 2016-10-20:
For ensure s the accuracy is just decrease speed of readings, try to change it to 250ksps instead 1msps

noisymime, 2016-10-20:
The sampling frequency has no impact on the time it takes to charge the sample capacitor, that is purely electrical, based on the impedance of the circuit being measured and the current being fed to the ADC.

tausday, 2016-10-21:
Hello noisymime,

i have test now the while loop in the main loop at 7560 RPM+-8RPM

Normal Loop = 860Loop/s
While Loop    = 870Loop/s

Now i go to bed

VitorBoss, 2016-10-21:
Yes, has no impact on the time to charge the capacitor, but impact the successive approximation circuitry.

`By default, the successive approximation circuitry requires an input clock frequency between 50kHz and 200kHz. If
a lower resolution than 10 bits is needed, the input clock frequency to the ADC can be as high as 1000kHz to get a
higher sample rate.`
ATmega640/V-1280/V-1281/V-2560/V-2561/V [DATASHEET] page 271
version 2549Q–AVR–02/2014

VitorBoss, 2017-02-06:
Sorry for long delay, the digitalWriteFast() function have a problem with interrupts, there is another version to  handle this but return an error wen compiling

noisymime, 2024-03-15:
Closing as stale

closed 2024-03-15 by noisymime