speeduino/speeduino #53: Analog readings

VitorBoss, 2016-10-25:
Using my last comment on issue #51 

> `By default, the successive approximation circuitry requires an input clock frequency between 50kHz and 200kHz. If a lower resolution than 10 bits is needed, the input clock frequency to the ADC can be as high as 1000kHz to get a higher sample rate.`
ATmega640/V-1280/V-1281/V-2560/V-2561/V [DATASHEET] page 271
version 2549Q–AVR–02/2014
Using the datasheet as base the ADC prescaler should be clk/128 to output a 125KHz clock os ADC successive approximation circuitry

I think is better to use analog interrupt handle. Make a struct named AnChannel with all 16 channels.

> ADMUX = 0x01000000; //Starting at channel 0
ADCSRA = 0x11101111; // ADC Interrupt Flag enabled and prescaler selected to clk/128
ADCSRB = 0x00000000; //ADC Auto Trigger Source is in Free Running mode
//ADCSRB = 0x00000100; //ADC Auto Trigger Source is Timer/Counter0 Overflow
//ADCSRB = 0x00000110; //ADC Auto Trigger Source is Timer/Counter1 Overflow
//
ISR (ADC_vect)
  {
  if((ADMUX & 0x00100000) //8 to 15
   { AnChannel(8+(ADMUX & 0x0000111)) = (ADCH<<8)+ADCL; }
  else
  {  AnChannel(ADMUX & 0x0000111) = (ADCH<<8)+ADCL; }
  if((ADMUX & 0x01100111) //channel 15
  {  ADMUX = 0x01000000; }//channel 0
  else if ((ADMUX & 0x01000111) //channel 7
  {  ADMUX = 0x01100000; }//channel 8
  else
  {  ADMUX++; }
  ADCSRA = 0x11101111; // ADC Interrupt Flag enabled and prescaler selected to clk/128
  } 
//Informations on pages 281 to 287 at 2549Q–AVR–02/2014 datasheet.

This will speedup the analog conversions, at prescaler selected to clk/128 will run at 125KHz(limit due to page 271), is 8ms continuous conversion interrupt.

noisymime, 2016-10-26:
The interrupt is without a doubt a more efficient way to do the readings, but consideration needs to be given to how the code can be shared on other platforms (Such as Teensy). Keep in mind also that doing it this way will require some form of ADC queue as there will be multiple analog reads requested concurrently.

For what it's worth, my testing shows no discernible difference in ADC accuracy at 1Mhz vs 125Khz. The definitive comment on this is included in AVR120 application note (http://www.atmel.com/Images/doc2559.pdf):

> The ADC accuracy also depends on the ADC clock. The recommended maximum ADC clock frequency is limited by the internal DAC in the conversion circuitry. For optimum performance, the ADC clock should not exceed 200 kHz. However, frequencies up to 1 MHz do not reduce the ADC resolution significantly.

Given that most of the analog values are scaled to 8 or 9 bits, the impact of a 1Mhz ADC clock is not noticeable.

VitorBoss, 2016-10-26:
Reading both Tennsy(3.5 and 3.6) datasheet, both have an interrupt mode for analog conversions, and both have a dedicate DMA option to do this with up to 16bits resolution. See on Chapter 39, Memory map and register definition: ADC Status and Control Registers 1 (ADCx_SC1n) section

([http://www.nxp.com/files/32bit/doc/ref_manual/K66P144M180SF5RMV2.pdf](url)) for 3.6
([http://www.nxp.com/files/microcontrollers/doc/ref_manual/K64P144M120SF5RM.pdf](url)) for 3.5

On Teensy datasheet don't seems to need a complex interrupt handle at all, and both have hardware average function and can run ADC clock frequency up to 4MHz.
If the impact at 1MHz is not noticeable keep it.

Maybe creating a header file to process all that?

jakemoroni, 2016-10-26:
VitorBoss, I think there are some issues with the example code you posted, but I agree with the general concept. I do realize you probably wrote that just to get your point across and that it’s not final code.

Basically, I feel that the analog inputs should be handled as follows:

-	Have conversions be initiated asynchronously and periodically (i.e., not initiated via the main program thread)
-	Use the “conversion complete” interrupt to grab the data from the respective channel and store it to some shared memory 
-	All “reads” of the channel would then be obtained by reading from the shared memory rather than the ADC data registers (no queue necessary, just a small critical region/barrier to ensure atomic access)

There are a few advantages to this approach, one of which being the fact that analog reads will now be deterministic in time (and also won’t block the main thread, assuming that the current code just uses the regular analogRead function). It will also lead to an overall performance increase since more things are happening in parallel. 

This is basically what I think you are getting at. However, one of the issues in your example is that new conversions are initiated automatically while in free running mode, and the value of the ADMUX register is basically “latched” one clock cycle before a conversion finishes. (See page 274).

Therefore, you will need to implement some extra logic to properly determine which channel the interrupt was for. 

Consider your example:

Before/on the first iteration, ADMUX is first set to channel 0, you kick off the sampling clock, then get an interrupt. This interrupt represents a completed sample for channel 0. Then, before exiting the ISR, you change ADMUX to channel 1. However, the next sample for channel 0 has already started (it was started automatically as soon as the last one finished). Therefore, the next interrupt that you receive will be for channel 0 _again_ since the value of ADMUX changed afterwards. However, you will mistakenly assume it to be channel 1 because you’re reading the ADMUX register to determine the channel. 

Basically, for all but the first interrupt, I think the interrupt will represent a completed sample for the channel of (ADMUX – 1) (wrapped).

Also note that there are some errors in how you’re determining the channel. The 6th MUX bit (MUX5) is actually in the ADCSRB register, not the ADMUX register. You’ll need to manipulate both of those. 

I haven’t actually tested this, so forgive me if I’m wrong. 

Thanks,
Jake

VitorBoss, 2016-10-27:
Yes Jake, it is a sketch, only to express my idea. I had read the related pages. Thank you for the MUX5 related mistake, I will try to fix that in following code.

ADCSRA register bit 4(ADIF) is messing with my head, on page 274 for a safely update says:

1. When ADATE or ADEN is cleared
2. During conversion, minimum one ADC clock cycle after the trigger event.
3. After a conversion, before the Interrupt Flag used as trigger source is cleared.

When updating ADMUX in one of these conditions, the new settings will affect the next ADC conversion.

But on page 285 says:
"ADIF is cleared by hardware when executing the corresponding interrupt handling vector. Alternatively, ADIF is cleared by writing a logical one to the flag."

> ADMUX = 0x01000000; //Starting at channel 0
ADCSRB = 0x00000000; //ADC Auto Trigger Source is in Free Running mode
//ADCSRB = 0x00000100; //ADC Auto Trigger Source is Timer/Counter0 Overflow
//ADCSRB = 0x00000110; //ADC Auto Trigger Source is Timer/Counter1 Overflow
ADCSRA = 0x11101100; // ADC Interrupt Flag enabled and prescaler selected to 1MHz
.
.
.
ISR (ADC_vect)
{
byte nChannel;
ADCSRA = 0x11001100; // ADC Auto Trigger disabled by clearing bit 5(ADATE)
nChannel = (ADMUX & 7);
if((ADCSRB & 0x00001000) { nChannel+=8; } //8 to 15
AnChannel(nChannel) = ADCL | (ADCH << 8);
if(nChannel==15) //channel 15
{ ADMUX = 0x01000000; } //channel 0
else if (nChannel==7) //channel 7
{ 
ADMUX = 0x01000000;
ADCSRB = 0x00001000; //Set MUX5 bit
}
else
{ ADMUX++; }
ADCSRA = 0x11101100; // ADC Interrupt Flag enabled
} 

Maybe work, I will try it later.

VitorBoss, 2016-11-03:
with this:
>  ADMUX = 0x40; //Starting at channel 0
  ADCSRB = 0x00; //ADC Auto Trigger Source is in Free Running mode
  //ADCSRB = 0x04; //ADC Auto Trigger Source is Timer/Counter0 Overflow
  //ADCSRB = 0x06; //ADC Auto Trigger Source is Timer/Counter1 Overflow
  ADCSRA = 0xED; // ADC Interrupt Flag enabled and prescaler selected to 500KHz
ISR(ADC_vect)
{
  byte nChannel;
  int result = ADCL | (ADCH << 8);  
  ADCcount++;
  ADCSRA = 0xCD; // ADC Auto Trigger disabled by clearing bit 5(ADATE)
  nChannel = ADMUX & 0x07;
  if(ADCSRB & 0x08) {nChannel+=8;} //8 to 15  
  if(nChannel==0) {AnChannel[15] = result;}
  else {AnChannel[nChannel-1] = result;}  
  if(nChannel==15)
  {
    ADMUX = 0x40; //channel 0
    ADCSRB = 0x00; //clear MUX5 bit
  }
  else if (nChannel==7) //channel 7
  { 
    ADMUX = 0x40;
    ADCSRB = 0x08; //Set MUX5 bit
  }
  else {ADMUX++;}  
  ADCSRA = 0xED; // ADC Interrupt Flag enabled
}

I got:
2500 Loops with Clean code and interrupts: 30704
 A0=963 A1=904 A2=844 A3=783 A4=723 A5=664 A6=602 A7=542 A8=481 A9=421 A10=360 A11=300 A12=240 A13=179 A14=119 A15=58
Number of analogic interrupts: 1179
 
This changing ADCSRA = 0xCC (prescaler selected to 1MHz) :

2500 Loops with Clean code and interrupts: 57164
 A0=930 A1=903 A2=843 A3=783 A4=723 A5=663 A6=601 A7=542 A8=481 A9=421 A10=360 A11=300 A12=240 A13=179 A14=119 A15=57
Number of analogic interrupts: 4395

and this changing bit ADCSRA = 0x6C and this code:

> //if(nChannel==0) {AnChannel[15] = result;}
  //else {AnChannel[nChannel-1] = result;}
  AnChannel[nChannel] = result;

2500 Loops with Clean code and interrupts: 27748
 A0=956 A1=897 A2=838 A3=777 A4=719 A5=660 A6=600 A7=542 A8=483 A9=425 A10=365 A11=305 A12=245 A13=185 A14=125 A15=63
Number of analogic interrupts: 880

VitorBoss, 2016-11-03:
PS. using 220R resistors from 5V to channel A0, 220R to A1 ... A15 to GND,

VitorBoss, 2016-11-05:
Codding done, now are working with main code.

closed 2016-11-05 by VitorBoss