speeduino/speeduino #66: General pinouts incorrect on the HC-01 and HC-02 and BoostOut pins

rocket777, 2017-02-07:
Hi all,

I recently connected the fuel pump and thermofan to the speeduino board V0.4.2. 
I tried to load the Dec16.zip firmware from the www.speeduino.com web page (outdated/different) and found that the firmware on the Wiki was different, especially the pin assignments in the Utils.ino file. 

I uploaded the firmware from the wiki, and started to trouble shoot why the Fuelpump and Thermofan wouldn't work when connected to the pins 14 and 15 on tue IDC40 connector. These ar slow current Relay switching pins, however these did absolutely nothing on the scope. I was then advised to use the HC (high current outputs 1 and 2) and did this by changing the Utils.ino file pins to read pinD4 and PinD5 (arduino pins) and then the same changes inside Tuner Studio - the thermofan i managed to get working correctly by setting it up on Pin5 inside Tuner Studio - 

The fuel pump however - wouldn't work. Josh then replied to my PM and told me to use PinD4 on the arduino to connect to IDC40 Pin38 - this is incorrect, D4 connects to IDC40 Pin36. 

I am now running the following successfully:- 

1) PinD4 connected to IDC40 Pin35 for the Fuel Pump output - Tuner studio pin set to pin7 --- 
2) PinD5 connected to IDC40 pin37 for the thermofan output - Tuner studio pin set to Pin5 ---

No matter what pin combination i use to get the boost control or VVT or Idle PWM or Idle Stepper to work, nothing will....... I have tried the IDC40 pins 34, 36 and 38 with no luck. I have checked the schematic off the WIKI and it says completely different pin connections versus Arduino pin connections. 

I would like to use the following features on my ECU but cannot seems to get these to work at all - 

1) Injectors (sequential 4cyl)
2) ignition sequential ( 4cyl) 
3) Thermo Fan
4) Fuel Pump
5) VVT
6) Idle control PWM
7) Idle control Stepper (second car)
8) Boost control
9) Launch control
10) O2 sensor 5wire wide band LSU4.9
11) Flex fuel

Can someone possibly shed some light on this??

rocket777, 2017-02-13:
Anyone?? has anyone possibly used a Speeduino to run an engine and actually try to use some of the features without having to modify code, and hardware?? Even so, please share mods or code changes to get this to work??

PSIG-dev, 2018-02-26:
There have been multiple threads about this, such as here (https://speeduino.com/forum/viewtopic.php?f=16&t=1623&p=23799&#p23799) and [here] (https://speeduino.com/forum/viewtopic.php?f=19&t=1023&p=16588&#p16588).  If testing, I suggest first with just the Mega (to avoid any issues with the PCB), and then with the board.  Use the schematics to verify the path a certain pin is outputting on; for example HC outputs are on D4, 5, 6 and 7, while LC is to the proto area and needs an output transistor, or ULN, and possibly jumpers to function.  Yes, code should probably default to an HC output for essential things like fuel pump, but the outputs should function, and are assignable.  Note all the users that have the functions you listed working. ;-)