speeduino/speeduino #70: Subaru 6/7 tooth trigger

noisymime, 2017-03-03:
https://speeduino.com/forum/viewtopic.php?f=12&t=80&p=16868&hilit=subaru#p16868

[image]

[image]

[image]

[image]

noisymime, 2017-03-31:
[image]
Adding version of the timing chart that's actually readable

Koop2k, 2017-08-01:
[image]
Thought this could be usable here.

mstevanin4, 2017-10-06:
According to my email to Josh on 19.07.2017, the feature was tested successfully. I think this issue can be closed.

noisymime, 2017-10-06:
Great, thanks for the reminder!

closed 2017-10-06 by noisymime

noisymime, 2018-11-21:
Composite log for reference: https://i.imgur.com/0aR4gIJ.png