speeduino/speeduino #76: 3000GT CAS Decoder

ElDominio, 2017-04-08:
http://speeduino.com/forum/viewtopic.php?p=17235#p17235

[image]

VitorBoss, 2017-04-09:
Are you planned this to be sequencial or wasted spark?

ElDominio, 2017-04-09:
While this gives full indication of engine position, Speeduino only has 4 injector/ignitor outputs, so wasted spark.

noisymime, 2017-04-09:
This appears to be basically the same as the 4g63 pattern, except with 6 inputs on the crank instead of 4. I may make the 4g63 pattern work with this if 6 cylinders are selected, or I might make a whole new decoder option, I'll have to see which makes the most sense. 

Either way, thanks for the diagram! That's exactly the sort of thing needed to write the decoder up :)

ElDominio, 2017-05-02:
No problem! I'm finally getting A/C in the shop and I'll probably conjure up one for a friend's 3000GT to start testing out on :)

noisymime, 2017-05-16:
Ready for testing whenever you are

ElDominio, 2018-04-22:
Testing will probably be done this week, will update on here when done!

ElDominio, 2018-06-15:
Nope, doesn't work! I've tried a bunch of things and nothing has made it get significantly better.

Here's my edited code

`void triggerSetup_4G63()
{
  triggerToothAngle = 180; //The number of degrees that passes from tooth to tooth (primary)
  toothCurrentCount = 99; //Fake tooth count represents no sync
  secondDerivEnabled = false;
  decoderIsSequential = true;
  decoderHasFixedCrankingTiming = true;
  triggerToothAngleIsCorrect = true;
  MAX_STALL_TIME = 366667UL; //Minimum 50rpm based on the 110 degree tooth spacing
  if(initialisationComplete == false) { toothLastToothTime = micros(); } //Set a startup value here to avoid filter errors when starting. This MUST have the initi check to prevent the fuel pump just staying on all the time
  //decoderIsLowRes = true;

  //Note that these angles are for every rising and falling edge
  if(configPage2.nCylinders == 6)
  {
    // 70 / 50 for 6 cylinder applications
    toothAngles[0] = 185; //
    toothAngles[1] = 235; //
    toothAngles[2] = 305; //
    toothAngles[3] = 355; //
    toothAngles[4] = 65; //
    toothAngles[5] = 115; //

    toothAngles[6] = 540; //
    toothAngles[7] = 595; //
    toothAngles[8] = 665; //
    toothAngles[9] = 715; //
    toothAngles[10] = 425; //
    toothAngles[11] = 475; //

    triggerActualTeeth = 6;
  }
  else
  {
    // 70 / 110 for 4 cylinder
    toothAngles[0] = 715; //Falling edge of tooth #1
    toothAngles[1] = 105; //Rising edge of tooth #2
    toothAngles[2] = 175; //Falling edge of tooth #2
    toothAngles[3] = 285; //Rising edge of tooth #1

    toothAngles[4] = 355; //Falling edge of tooth #1
    toothAngles[5] = 465; //Rising edge of tooth #2
    toothAngles[6] = 535; //Falling edge of tooth #2
    toothAngles[7] = 645; //Rising edge of tooth #1

    triggerActualTeeth = 4;
  }
  /*
   * https://forums.libreems.org/attachment.php?aid=34
  toothAngles[0] = 715; //Falling edge of tooth #1
  toothAngles[1] = 49;  //Falling edge of wide cam
  toothAngles[2] = 105; //Rising edge of tooth #2
  toothAngles[3] = 175; //Falling edge of tooth #2
  toothAngles[4] = 229; //Rising edge of narrow cam tooth (??)
  toothAngles[5] = 285; //Rising edge of tooth #3
  toothAngles[6] = 319; //Falling edge of narrow cam tooth
  toothAngles[7] = 355; //falling edge of tooth #3
  toothAngles[8] = 465; //Rising edge of tooth #4
  toothAngles[9] = 535; //Falling edge of tooth #4
  toothAngles[10] = 535; //Rising edge of wide cam tooth
  toothAngles[11] = 645; //Rising edge of tooth #1
   */

  triggerFilterTime = 1500; //10000 rpm, assuming we're triggering on both edges off the crank tooth.
  triggerSecFilterTime = (int)(1000000 / (MAX_RPM / 60 * 2)) / 2; //Same as above, but fixed at 2 teeth on the secondary input and divided by 2 (for cam speed)
  triggerSecFilterTime_duration = 4000;
[... 259 lines cut ...]
      interrupts();

      crankAngle = toothAngles[(tempToothCurrentCount - 1)] + configPage4.triggerAngle; //Perform a lookup of the fixed toothAngles array to find what the angle of the last tooth passed was.

      //Estimate the number of degrees travelled since the last tooth}
      //crankAngle += uSToDegrees(elapsedTime);
      unsigned long toothTime = tempToothLastToothTime - tempToothLastMinusOneToothTime;
      crankAngle += int((elapsedTime * triggerToothAngle) / toothTime);
      //timePerDegree = toothTime / tempToothAngle;

      if (crankAngle >= 720) { crankAngle -= 720; }
      if (crankAngle > CRANK_ANGLE_MAX) { crankAngle -= CRANK_ANGLE_MAX; }
      if (crankAngle < 0) { crankAngle += 360; }
    }
    return crankAngle;
}

void triggerSetEndTeeth_4G63()
{
  if((ignition1EndAngle - configPage4.triggerAngle) > 355) { ignition1EndTooth = 1; }
  else { ignition1EndTooth = 4; }
  //ignition1EndTooth = 1;
  //ignition1EndTooth = ( (ignition1EndAngle - configPage4.triggerAngle) /
}
`

noisymime, 2018-06-17:
This is ready for testing now with the latest commit. Seems to be working on the bench for all known 6 cylinder configs.