speeduino/speeduino #81: Won't detect trigger

apollard, 2017-04-30:
I've been having trouble getting Speedy to recognize the basic distributor trigger. 

If I turn the key and immediately crank the engine, it always detects the trigger. 

If I turn the key on and wait a few seconds, it never detects the trigger. 

Never fails, I tried it probably 40 times back and forth. Then cranked it immediately upon key turn 10 more times and everytime it synced. Video below shows this clearly. For this video, recall that I am only trying to get the trigger to work reliably, the distributor and carb are running the engine.

First I turn the key and crank it. You can see the injector lights flashing, showing trigger sync. Second, I turn the key, wait a few seconds and crank it. You can hear the FP relay cycle for the priming pulse and then when I start the engine you can hear the FP relay clicking. No injector lights, and TS reports no teeth in the logger and no RPM.

I do this a couple more times in the video to show the pattern.

https://youtu.be/owCFiOiRrDU

Pretty sure this isn't expected behavior, hence the report.

Original thread here: http://speeduino.com/forum/viewtopic.php?f=19&t=1140&p=18277#p18277

apollard, 2017-05-02:
Started trying to figure out where in the code it is causing the FP to cycle on/off (ie, relay clicking) and more definition of exact conditions causing it. Put in serial prints to find area that is cycling pump.

apollard, 2017-05-03:
OK, so did some more testing, and I've narrowed this issue down to just the "basic Distributor" trigger. 

Speedy works fine with any trigger other than those. You can delay the trigger signal as long as you like and it syncs fine with other wheels. You can also remove the trigger signal, then reapply it and Speedy will sync again. 

With the basic distributor setting, the trigger must be there when Speedy boots up or it won't sync and it goes into a situation where it pulses the Fuel Pump approximately every second. If the trigger is removed, it won't re-sync when it is reapplied.

Now to figure out what is different about the Basic Distributor code that causes this.

noisymime, 2017-05-04:
Thanks for the report and the investigation you've done into this!

I'm not immediately seeing this on a test setup, but will keep looking into it.

noisymime, 2017-05-04:
Can you please post a tune file (msq) for your setup?

DarrenInwood, 2017-05-04:
That certainly sounds like a restart of the CPU to me. Is your fuel pump
pulling too much current, which drops the supply voltage too low, which
then restarts the whole shebang ad infinitum?

noisymime, 2017-05-04:
@apollard Does TS reconnect / disconnect when the problem occurs? If it is then the resetting mentioned by @DarrenInwood may be a consideration. If TS stays connected, then that can be rules out.

I pushed up a few small fixes just now. Would be worth trying them to see if they make any difference, but I'm still unable to reproduce the issue the way you're seeing it, so I wouldn't be confident they have fixed it.

apollard, 2017-05-04:
"fix Cranking Timing With Trigger" made no Difference.

However, with the Ardustim attached, I decided to change the filter I had (Weak) to none. That DOES show teeth in the logger. Below is the pattern I'm seeing (Tooth logger with NO sync - 8 cyl dizzy). 

Since the Ardustim pattern is a 50% duty cycle square wave, the timing on each tooth should be very close, as shown by the image with sync (Tooth logger with sync - 8 cyl dizzy). The only change between these two images is that I power cycled the Speedy. Ardustim running with no change, so trigger was there on Speedy boot.

So something in code is throwing the timing of the trigger interrupt off ?
[image]
[image]

apollard, 2017-05-04:
TS is not dropping connection, and the arduino stays operating (ie, not rebooting). Right now, I only have the FP relay hooked up in the boat, but not the FP. It isn't loosing power, since the ECU and FP relay are fed from different home runs to the main power point for the boat.

I am now simulating this on the bench with just an Mega 2560, a LED/resistor to show FP state and an Ardustim for trigger. So, no load at all really.

Current tune below - this is the one generating tooth logs above.

I'll download the commit and try it.

[CurrentTune.zip](https://github.com/noisymime/speeduino/files/976360/CurrentTune.zip)

noisymime, 2017-05-04:
Hmmmm interesting. Almost certainly something to do with the filtering then based on your last 2 comments (and what I changed today). Anyway, glad to hear it's progress. 

Happy to close this one or do you want to do more testing?

apollard, 2017-05-04:
I think you can close it. I was able to replicate on the bench so reliably that I'm 100% sure it will work on the boat.

noisymime, 2017-05-04:
Fixed by bae708a4d638aae40ccf0978b7a741d994717f82

closed 2017-05-04 by noisymime