speeduino/speeduino #101: Rover MEMS pattern

noisymime, 2017-10-10:
Per thread: https://speeduino.com/forum/viewtopic.php?p=21572&sid=e3841610b755204f35a86d9cf3f43041#p21572

32 poles spaced at 10 degree intervals, with 4 missing poles,at 30, 60,210 and 250 degrees.

[image]

Autohome2, 2017-11-17:
further information on this trigger pattern
[mems3 crank and cam.pdf](https://github.com/noisymime/speeduino/files/1481867/mems3.crank.and.cam.pdf)

andymcg85, 2018-03-18:
Following for updates.

yeco, 2018-03-19:
Following.

mapgdb, 2020-06-09:
Any progress? Looking into starting my own Speeduino project and this would help out!

mike501, 2020-10-27:
Rover have multiple mems trigger patterns from what I can gather. I'm aware of 3 and their maybe a 4th!
So far I've got,
Pattern 1 36-1-1        (17 teeth - gap - 17 teeth - gap)
Pattern 2 36-1-1-1-1 ( 13 teeth - gap - 2 teeth - gap - 14 teeth - gap - 3 teeth - gap) 
Pattern 3 36-1-1-1-1 ( 11 teeth - gap - 5 teeth - gap - 12 teeth - gap - 4 teeth - gap)

The 'maybe' pattern i can't find anymore but it think it was a variation on mode 2 & am hopeful it was someone looking at the same wheel differently.

I've asked the person who said they had a working pattern to post up their details. If it works, i'll steal it to make the alternate pattern. If they don't post i'll look to create the patterns.

Pattern 1 is really a 18 - 1 at double crank speed. I'm curious if you could get away saying its the cam signal intead of crank.
Pattern 2 and 3 i'm wondering if I could 'cheat' and count teeth. Once I find (pattern 2) 14 teeth or (pattern 3) 12 teeth i know where i am on the crank. 

I don't have details on the cam sensor and its relation to the crank - hence can't do that.
I've attached the spreadsheet i've created with the teeth mappings in for ardustim.
[rover mems trigger pattern.xlsx](https://github.com/noisymime/speeduino/files/5445299/rover.mems.trigger.pattern.xlsx)

james-portman, 2020-11-08:
Clearer picture below of the original flywheel the creator is talking about, "Pattern 2" as Mike above has written about.

It would be good to get pattern 1 & 2 working as above for the Rover Minis to provide a cheaper alternative to current ECU swaps.
I don't mind having a go at this if nobody else gets it working first.

I think..
Pattern 1 - for cars with (ECU timed) distributors to send the spark going to the right cylinder, and single point injection.
Pattern 2 - for cars with wasted spark coil packs. Single point injection without a cam sensor, can be multi point with a cam sensor.
Pattern 3 - same as 2

[image]

andymcg85, 2020-11-08:
I did a rover k series turbo conversion and I don't know if it's a simpler
option altho it was for me as it was several years ago and no body wanted
to create the trigger pattern. I welded a 36th wheel to the back of crank
pulley and it worked a treat

james-portman, 2020-11-08:
Can anyone direct me a bit?
If I make new code then I have a Mega here I can test it on, I got that up and running in tunerstudio.
If I get another arduino to pretend to be the crank signal, which pin on the mega is it?

I tried to look through the init code but it gets a bit wild..

james-portman, 2020-11-09:
Started code (without cam support) here:
https://github.com/noisymime/speeduino/compare/master...james-portman:rover-36-1-1-1-jamesp?expand=1
Working for JDM Mini SPI flywheel pattern perfectly.

Could be possible to adapt slightly to fit many cars with multiple gap pattern flywheels, at least the other Rover flywheels and add cam support

commit 53d6a6a4 (2023-04-30): Rover MEMS 4 cylinder decoder (#1032)

Decodes the 5 known Rover MEMS ECU patterns for Rover based engines. Covers A series, T series, O Series, K Series and V8. Doesn't currently cover sequential ignition for v6 or v8.

Resolves #101
closed 2023-04-30 by noisymime