speeduino/speeduino #107: Renix Decoder(44-2-2/66-2-2-2)

AKCore, 2017-10-25:
Decoder for Early Jeeps some volvos and renaluts with Renix MPFI. 
This trigger mode came in a four cylinder variation which used 44 base teeth with two gaps 180
degrees apart, and a six cylinder version with 66 base teeth and three gaps 120 degrees apart. 
[image]

romainbavera, 2019-04-11:
Hi,
No solution?

giardadiego, 2020-09-23:
Hi! Many Renault clio are waiting for it😆. Any possibility to implement it?

xx75ccy, 2020-11-08:
hi guys, i'm new user form italy! i have to be onest, i'm registered to partecipate at this tread.
i'm interested in renix 44-2-2 wheel decoder, and i'm going to attached some info of this.
olso i have a renault engine f7p on the floor, so i can put other information if you need.
thanks in advance! :D :D
[image]
[image]
[image]

tefnet77, 2021-04-11:
I would love to put an speeduino in my jeep if the renix 66-2-2-2 is decoded

mike501, 2021-04-11:
Does this engine also have a cam signal or does it still use the distributor for ignition? I'm asking because the pattern above isn't unique in a 360 degree cycle.

tefnet77, 2021-04-11:
the motor have a cam sensor in the distributor.
i hope this page has all the info you ask.
https://jeep-manual.ru/index.php?page=134

mike501, 2021-04-11:
Fab, that covers the 6 cylinder Jeep. Any details on the 4 cylinder Renault or do we expect it to be similar (cam trailing edge on the last cylinder)

xx75ccy, 2021-04-12:
hi guys, i'm interested to in this pattern to put in a old renault clio with the 44-2-2 renix wheel. renault don't have any sync signal, it work only in simultaneus injection and 1 coil with distributor. i'm sure the megasquirt have this pattern in his code, maybe it can be useful to look at, and it can work on simultaneus injection like the original ecu and it can do full sequential with addition of cam sensor. if you need more info i have an engine on the floor

tefnet77, 2021-04-12:
the jeep renix 2.5 (4 cylinder) has the info in this page about the flywheel http://jeep-manual.ru/index.php?page=126
maybe xx75ccy can send a picture of the flywheel to see if it matches that description.

xx75ccy, 2021-04-13:
Hi, this Is the Renault 44-2-2 flywheel. The couple of missing tooth/add.tooth are at 180° from each other. Tomorrow i'll mesure the atdc angle 
[image]
[image]

DrFiamma, 2021-04-15:
Hi everybody, I am very interested too about this topic
I am from Italy, with a Renault engine trying to get speeduino working with a 44-2-2 flywheel

Any news about it?

tefnet77, 2021-04-15:
i have found another page with information about crank and cam signal of the 66-2-2-2 
http://www.ajeepear.com/docs4x4/Jeep_Renix_Fuel_Injection_manual.pdf
it starts on page 20

mike501, 2021-04-15:
I'm curious about the size of the teeth and size of the gaps. 44 divided by 360 is 8.181818. 44 times 0.181818 is 8. Makes me wonder if the -2 tooth size is actually bigger so the maths would be easy with each tooth being 8 degrees instead of fractionally larger.

xx75ccy, 2021-04-17:
You're in the right, i'm going to verify this asap and report...

tefnet77, 2021-04-17:
i have found more information about the 66-2-2-2
source: http://jeep.avtograd.ru/cherokee/injection/89jeepefi.pdf
[image]

noisymime, 2021-04-17:
> I'm curious about the size of the teeth and size of the gaps. 44 divided by 360 is 8.181818. 44 times 0.181818 is 8. Makes me wonder if the -2 tooth size is actually bigger so the maths would be easy with each tooth being 8 degrees instead of fractionally larger.

Would be really great if we can confirm this, although it will be a tricky one to physically measure as it's a fairly small difference. 

Looking around at other implementations, they appear to use the 8.1818 value, but that's not to say they're correct.

mike501, 2021-04-18:
My thought was partly prompted by a picture of a scope I saw on the web which i swear showed a gap a bit larger than 2 teeth.

I was hoping that if the flywheel was (guessed) 30cm ISH that would make the 0.181818 about 0.5mm which should be measurable on a digital caliper. I hadn't thought someone might try using a ruler. However it would make the -2-2 gap easily measurable of it was 4 or 5 teeth in size.

tefnet77, 2021-04-18:
I think you are right, it seems that the gap is bigger than 2 teeth (at least in the 66-2-2-2)
here is a video of a guy changing the flywheel where you can see the gap
https://youtu.be/CgUsJi0Jq2U?t=738
more like a 2,5 teeth gap?

xx75ccy, 2021-04-18:
Today i took some measures and also took some pics of these. How you can see the theet have an involute shape, so not very easy to take the perfect measure, but i have try to take them with the max precision as possible. 
[image]
[image]
[image]
[image]
[image]
[image]

How you can see, the lenght of the +2 theet is different from the -2 teeth

At the end i take the atcd point. the +2 teets start at 114° before tdc 1/4 cilinder, and the -2 theet pass the sensor at 80° before tdc. 
[image]
[image]

hopeing this will help someone...
bye

mike501, 2021-04-18:
Fabulous, thank you.

I think the numbers prove my theory was wrong, it does use a strange angle between teeth.

Over to Josh to work his magic with the numbers.

ptitbobengt, 2021-06-03:
I am attaching an extract from a technical document
on the gt turbo

Le volant moteur:

Basiquement le capteur détecte les dents du volant moteur, elle sont au nombre de 40 dont deux sont plus larges que les autres
et à 180° l'une de l'autre. Le calculateur met en forme les signaux pour reconnaitre le PMH, le PMB et le régime moteur.
Ci-dessous un dessin du volant moteur.

Dents du volant moteur.

[image]

ptitbobengt, 2021-06-04:
I found this link

can it be used?

sorry for my english

https://wiki.speeduino.com/en/Trigger_Wheel_Generator

Autohome2, 2021-06-04:
Can it be used for what?
That link is to the speeduino wiki for the wheel generator for fab work,
nothing to do with firmware if that was your thoughts?

mike501, 2021-06-23:
Had a random thought about this pattern yesterday. I think every few teeth the pattern lines up with a proper angle. Eg tooth 0 is 0 degrees, tooth 5 is 40.909090 degrees (41 degrees), tooth 11 is 90 degrees, tooth 16 is 130.909090 (131) degrees etc etc. I don't have the time but I can see a bit of code where you system count the actual teeth and every X teeth you increment the current tooth which maps to an array of tooth angles like other patterns use. You'd then end up with an 8 tooth pattern.

(That makes perfect sense in my head)

mike501, 2021-08-04:
I've started writing my version suggested above

xx75ccy, 2021-08-05:
> I've started writing my version suggested above for the 44-2-2 version
 Wow, amazing! We can't wait for It🤩🤩🤩

mike501, 2021-08-15:
https://github.com/mike501/speeduino/tree/Renix44-2-2

It might says 44-2-2 but i've also coded 66-2-2-2. Will need testing as i had to guess at the relationship between crank and cam (so i'll probably need to update that - if someone could get a tooth logger & composite logger that would be fab.)

tefnet77, 2021-08-16:
I will test the 66-2-2-2 in a few days, i will tell you something then

tefnet77, 2021-08-18:
I have compiled the software with arduino and vscode but i have problem selecting "Renix 66-2-2-2 - 6Cyl" it keeps saying "Burn page failed CRC Check"
with "Renix 44-2-2 - 4Cyl" i have no problem. 
the log is attached, 
is anything i am doing wrong or is a problem in the code?
[TunerStudioAppDebug.txt](https://github.com/noisymime/speeduino/files/7009689/TunerStudioAppDebug.txt)

debrucer, 2021-08-18:
Do you actually have a port 3 as it seems to have failed to communicate a
number of times?

August release is available now, too.

debrucer, 2021-08-18:
I suggested that based on thinking about Windows 10. Sometimes the port
settings get Maked "in use" and you don't see it. Can't say on any other
OS. Here's were to look for the setting
[image: port-settings.PNG]

Someone who knows the code and the logs better than I do may correct me,
but it is entirely possible that the repeated failure at the beginning of
the log could reflect on the final portion where something gets written. It
appears to work in the middle, but fails to produce the expected results.
I'd look at my ports.

mike501, 2021-08-19:
Will try and have a look later but may be Friday. I did a code update to the latest firmware with 66 already selected then moved to 44 tooth so possibly a glitch slipped in (aka I messed up)

mike501, 2021-08-20:
I certainly did mess up with rushing. Seems I did a = instead of a ==. 

Testing the changes now and will hopefully get something loaded later tonight.

mike501, 2021-08-23:
I'm being picky with my testing and may have found something in the general firmware so this will take a bit longer. If someone could get the tooth loggers (especially composite) that would be really appreciated as I can double check at the same time my assumptions on the relationship with the crank and cam.

xx75ccy, 2021-08-23:
I'm Sorry but at the Moment i can't do this, if none can do the log in this days, o can do It in next months (speacking of 44-2-2)...

mike501, 2021-10-19:
Sorry for the lack of updates. Found some stuff with the firmware which distracted me. Got 66 tooth pattern working except for an occasional short dwell on a single spark channel.  Once figured then migrating to 44 tooth should be a few days as the code code is written. I'm aware of a few changes proposed on ignition which I'll check to see if they help the dwell problem.

As for inclusion in the main branch, that will need some people to prove in a real car once the code is completed.

mike501, 2021-11-04:
Code is written and covers both 66 and 44 tooth cranks. I've had to guess at the cam relationship to the crank (ie it will be wrong but i needed something for ardustim). If you could provide the crank / cam relationship that would be fab.

tefnet77, 2021-11-13:
I have an issue with my board, when I get that fixed I'll try your code.

tefnet77, 2021-12-08:
I have tried today the 66-2-2 but the rpm keeps up and down and the motor barely keeps running during a few seconds.
There is attached a composite log with these few seconds could keep running.
I have tried with an ardustim that I have created and it happens the same, the rpm keeps up and down, if I disconnect the cam sensor the rpm reads correctly.

There is a problem with the fuel pin as it keeps up all the time, I have tested with two boards. With the official release the pin goes down after the prime duration.

[2021-12-08_13.08.11.csv](https://github.com/noisymime/speeduino/files/7677119/2021-12-08_13.08.11.csv)

mike501, 2021-12-08:
Not sure what the fuel pin problem is, will look at that once I have corrected the details on the relationship between crank and cam. 

Could you send me a copy of your ardustim & how did you establish when the cam started/ended in the cycle? I'll update the decoder to use the details you've established for ardustim if you're confident they're correct. What trigger offset did you use?

I've looked at the log file but it's not run for long enough to give me the information I need hence the interest in your ardustim setup. Alternatively if you could spin the engine with no spark plugs & no fuel we should be able to get a good 10 second log file.

tefnet77, 2021-12-10:
here is my ardustim and a few logs.

The ardustim is based in the picture I posted here on 17 Apr, I’m not sure is correct I only tried to create one that fit that pattern.

I have a problem with tunerstudio, most of the time the logs it creates does not have the first lines describing the info inside the log so the megalogviewer says “the file contains no data or is not recognized” I have to add manually this information to the logs to make it readably. One of the logs in the zip is this way, I haven’t add that info.

I don’t understand when you ask (What trigger offset did you use?) in ardustim? I think there is no offset to set in ardustim, if you are asking about tunerstudio I set to the trigger angle to 15º but I’m not sure is correct as the strobe gun sometimes shows the mark and sometimes not with this firmware.

[logs.zip](https://github.com/noisymime/speeduino/files/7694933/logs.zip)

[ardu-stim-master.zip](https://github.com/noisymime/speeduino/files/7694972/ardu-stim-master.zip)

mike501, 2021-12-11:
Code updated to run with your ardu-stim setup. Enabled sequential ignition and fuel without problem on the test rig.

https://github.com/mike501/speeduino/tree/Renix44-2-2

xx75ccy, 2022-01-02:
hi guys, someone had tested the code? What cam/crank relation was write in the firmware at the end ? Cause originally the 44-2-2 don't have the cam signal, it's an implementation.

mike501, 2022-01-02:
Code was updated to remove cam option for 44 tooth & I got the detail on
the cam position for 66 tooth so updated that.

Not heard back from the bloke testing the 44 tooth code. Will chase him now.

Can't remember who confirmed the 66 tooth cam position, will hunt them down
and see if they've had any success.

xx75ccy, 2022-01-03:
ok, i wish in the future will be a 44thooth with an additional cam sensor, now i run the 44-2-2 with a cam sensor on a MS2, but i would like to try speedy in future

mike501, 2022-01-03:
I still have my older version of the code in github. Happy for you to try
it out. Raise any issues on my github so we keep this thread clear till
we're at a working version.

tefnet77, 2022-01-09:
Sorry I haven’t post earlier.
I tested this morning the last speeduino firmware and I was able to get a much better composite log as it has no gaps.
[2022-01-09_13.30.18.csv](https://github.com/noisymime/speeduino/files/7835115/2022-01-09_13.30.18.csv)

With your firmware I was unable to get the renix firmware running, the code you posted on November barely keept the motor running with a lot of misfires, the one you modify with my ardustim wont start the motor at all.

mike501, 2022-01-09:
Will look at the composite log properly later. Can see from a quick look a little noise occasionally.  Setting the filter to low will help with that. Can you confirm what you did after re-loading the code - eg did you re-checked ignition timing offset with a timing light? Can you also send a copy of your tune?

I don't think either of these are the problem as it seems to lose sync every 19 teeth (20 if it has noise). Its highly likely to be the point the cam signal arrives doesn't quite match with what I'm expecting. If I can get a few hours looking at the logs I'll hopefully be able to correct this.

Conscious lots of people follow this board and we're likely to bounce a few messages back and forth working through this - do you want to log an issue on my github where you downloaded the code and we'll work this out between us.

tefnet77, 2022-01-10:
it seems that you have disabled the issues in your github , can you enable it o propose another way of communication?

mike501, 2022-01-10:
Try again, I've enabled it now.

mike501, 2022-01-18:
@xx75ccy - could you please measure the outside diameter of your flywheel? I'm expecting you to get either 252mm or 263.5mm. I want the size if all the teeth were solid (so if you end up measuring to a gap imagine the gap was a tooth)

Something about your measurements has concerned me and it's finally clicked. The -2 "gap" consisting large tooth and large space doesn't fit in with the measurements for the 44 tooth wheel. They equate to 3 teeth in actual size which means all my calculations may be out for both patterns :(

tefnet77, 2022-01-18:
I don't have the measurements of the 66-2-2 but in this video is seems that the -2 gap is more like -2.5 in size

https://youtu.be/CgUsJi0Jq2U

linkuso, 2022-01-18:
I dont know if it helps but 
[image]
I have a 44 2 2 wheel and if my math an meassurements are out of whack its 252mm in diameter, the small teeths are 11mm, spacing between the small teeth are 12mm, wide teeth is 27mm and the wide spacing is also 27mm.

xx75ccy, 2022-01-18:
> @xx75ccy - could you please measure the outside diameter of your flywheel? I'm expecting you to get either 252mm or 263.5mm. I want the size if all the teeth were solid (so if you end up measuring [...]

certainly, tomorrow morning i'll take the mesure!

mike501, 2022-03-03:
Any update on the measurement?

mike501, 2022-03-08:
Thanks for the updates xx75ccy & linkuso. Long post warning - trying to explain my logic and ask for someone to tell me what is wrong.

Both of you have provided details that say the normal tooth plus gap is 18mm and both say 54mm for the large tooth and large gap making up the -2 in the pattern description or "double tooth gap".

Two times the single tooth is 36mm & this is what my code assumed is the size of the large tooth gap based on it being described as -2, therefore my suspicion something didn't add up was correct. Three times the normal tooth is 54mm which matches your large tooth size. The sizes imply that the pattern should potentially really be described as 46-3-3.

However, something else doesn't add up. Pattern is 40-2-2 so 40 normal teeth + large tooth gap + large tooth gap.

40 normal teeth is 40 * 18mm =  720mm, add two lots of a large tooth gap 54+54 = 108mm and you get a total circumference of 828mm. 

You've both provided diameters within 2mm. To calculate circumference you do diameter x pi = and substituting your diameters you get 791 or 797mm circumference. Approx 36mm different to what it should be from the tooth measurements.

Either your diameter measurement didn't measure the same point you measured the teeth size or something else is wrong. Quick maths says we need the outside diameter to be 264mm to get a circumference of 829.3mm to get close to matching the teeth sizes. It gives an indication we need to find in the diameter approx. 12mm (or 6mm each side) to get the correct calculated circumference - is that the height of a tooth? 

Weirdly, If the large gap was 36mm instead of 54mm - ie exactly double the size of a single tooth + gap, then we'd get a circumference of 792mm. Which is within 0.4mm of what linkuso measured. Except we have a picture showing that isn't the measurement.

So in summary - we need to explain why measurements don't add up, until we do the code does not work as it stands. Using the measurements we have it can't work. Previously I'd calculated 360/44 teeth = 8.1818 degrees. Every x teeth it aligned with a whole number, I counted the teeth and let the system see a tooth when X number had passed. With the current sizes indicating 46 teeth we never get a whole number except at 0 and 180 degrees.

xx75ccy, 2022-03-08:
trying to thinking about this, i'm not sure if this is really the trick but try to explain my think. you consider the missing tooth as one 54mm long tooth, but if you watch the pics, after the 2 long teeth ( 27.5mm + 26.4mm) there is another normal 6.4mm theet directly attached to the second long theet (the 26.4mm long) whitout space from it. 
So i think you can't simply consider the 54mm as one long tooth because in reality you find a rougly 60.5mm long tooth (54+6.4mm). 

repeat, not sure about this reasoning....

mike501, 2022-03-09:
When I worked this out on a spreadsheet I listed every tooth & gap with their size. Depending which way the flywheel spins depends what you see but the spreadsheet had everything in it.

xx75ccy, 2022-03-09:
today i retake the diameter, this time with the right tool, still 254mm
![1646857113460](https://user-images.githubusercontent.com/74151361/157530674-95b42f88-4a9f-4870-8e74-53747e67a3ed.jpg
[image]
)

mike501, 2022-03-10:
Think I might have figured it, I accidentally started counting teeth not tooth  edges. Will test in the coming days but we could be ok

[6 more replies]