speeduino/speeduino #117: Toyota 36-2 + 3 Decoder

ElDominio, 2017-11-28:
Decoder for newer Toyotas, pending waveforms

ElDominio, 2017-11-29:
[https://drive.google.com/open?id=1ZSCGGtc3eM9peHsCo4XzhE-S3thYAnit](url)

ElDominio, 2017-11-29:
[image]

xalejandro1, 2017-12-14:
hey guys, I have a 2ZZ already running, semi sequentially. I'd like support for this decoder so it can be run fully seuqntial, and I'm willing to test this code.

ConnerMcLaughlin, 2017-12-21:
I made a pull request for my speeduino fork for completed code for this trigger wheel here: #120  Really the trigger wheel should be called 36-2/4-1 not 36-2+3 because it is essentially a missing tooth type trigger on primary with a missing tooth type trigger on secondary.  In other words it's a missing tooth dual wheel type.  How complete the code is is determined by the feedback of the testers.

ConnerMcLaughlin, 2017-12-21:
...Oh to set it up you need to pick Missing tooth in the trigger pattern with 36 teeth and 2 missing then the extra setting of "Missing Tooth Secondary type" as 4-1 cam.  The rest of the setting the testers need to figure out.

ConnerMcLaughlin, 2017-12-23:
https://github.com/ConnerMcLaughlin/speeduino/tree/36-2+3

ElDominio, 2017-12-23:
Issue is resolved thanks to Conner. Works great on a Corolla XRS.

noisymime, 2018-01-27:
PR #120 has been merged and adds this functionality.

closed 2018-01-27 by noisymime