speeduino/speeduino #130: Triggering for VW AEX engine.

mpex2006km, 2018-02-26:
I want to convert my car from the original Bosch MP 9 ecu to speeduino.  The only problem that I seem to have is that the OEM trigger design has an Optical sensor on the distributor with four teeth. One tooth is 6 degrees longer and that marks the #1 cylinder and to be exact the 72 degrees BTDC.

jrbe, 2019-07-06:
Are you sure is optical? VAG typically uses hall sensors.
If the one longer trigger window has the same edge spacing as the other trigger windows you should be able to set speeduino to watch for the rising or falling edge (whichever it is) and out should ignore the extra long part of the window. You will likely also have to offset the timing to match.
Other option would be to try a distributor trigger wheel from other vws, maybe the mid 90s 16v?

960, 2019-08-26:
I am looking for the VAG VVT decoder(4 Flanks) as well.

In reference position(TDC) the flanks should be:
1 = 104
2 = 284 
3 = 464 
4=  644
 
http://files.s4wiki.com/docs/Audi_R4-5V_T_132kW_ME7.pdf      Page 798

Med 9 ecus have the same pattern:

http://files.s4wiki.com/docs/MED9.1%20TFSI.pdf