speeduino/speeduino #131: Reverse option for TPS

PSIG-dev, 2018-02-26:
As previously discussed on Slack for a "reverse" checkbox in TS, and one code solution posted and tested here <https://speeduino.com/forum/viewtopic.php?f=19&t=1159#p18146>.

VitorBoss, 2018-03-03:
That was discussed almost a year ago. I had forgot about this

ConnerMcLaughlin, 2018-06-13:
Need some testers for this.
https://github.com/ConnerMcLaughlin/speeduino/tree/revtps

noisymime, 2018-06-17:
I've spent a bit of time playing with various options for this and have decided on an auto switching one. Due to the way the calibration works in TS, I cannot add a checkbox or dropdown to the calibration dialog, which is really the only place it makes sense to have it. 

In order to not confuse people by burying an option somewhere it may not be found, I think auto switching when the values are 'backwards' is best. The user will still get the warning, but the TPS readings will then be correct. 

Commit incoming to close this issue

commit 600fa72f (2018-06-17): Auto switching of TPS reverse calibration. Closes #131
closed 2018-06-17 by noisymime