speeduino/speeduino #138: Easy: RPM-dependent TPSdot AE

DAmx-5, 2018-04-23:
Looking to scale down AE as RPM increases, similar to feature in MS. At low RPMs, higher AE is needed than at high RPM. Currently I can only either start from a stop correctly (with no lean spot on throttle open) but go pig-rich/sputter when accelerating while in-gear or vice versa.

DAmx-5, 2018-04-23:
Here's what I did in the meantime:

 //If TAE isn't currently turned on, need to check whether it needs to be turned on
      int rateOfChange = ldiv(1000000, (currentStatus.TPS_time - currentStatus.TPSlast_time)).quot * TPS_change; //This is the % per second that the TPS has moved
      currentStatus.tpsDOT = rateOfChange / 10; //The TAE bins are divided by 10 in order to allow them to be stored in a byte. Faster as this than divu10

      if ((rateOfChange > configPage2.tpsThresh) **&& (currentStatus.RPM < 4000)**)
      {
        BIT_SET(currentStatus.engine, BIT_ENGINE_ACC); //Mark accleration enrichment as active.
        currentStatus.TAEEndTime = micros_safe() + ((unsigned long)configPage2.taeTime * 10000); //Set the time in the future where the enrichment will be turned off. taeTime is stored as mS / 10, so multiply it by 100 to get it in uS
        accelValue = 100 + (table2D_getValue(&taeTable, currentStatus.tpsDOT))***(1-(currentStatus.RPM)/4000)**;

lsdlsd88, 2018-04-29:
any plan for EAE (Enhanced Acceleration Enrichment, wall wetting and so on)?
TPS based is really not up to the task imho.

noisymime, 2019-01-22:
Closing as this was implemented in 3773647 and then fixed to work properly in 5d271ef

closed 2019-01-22 by noisymime