speeduino/speeduino #145: MAP based cycle sensing/phasin

Boogmeister, 2018-06-23:
https://speeduino.com/forum/viewtopic.php?f=18&t=1695&p=25451&hilit=MAP+phasing#p25451

Tobbera, 2018-07-11:
### MAP phase detection for multi cylinder engines
I'm very interested in this feature. Its important that it will work on more than two cylinder engines, with one parallel MAP sensor. As far as I know, the code needs to only watch for drop in pressure during certain time/rotation windows to make this work. As an example, my three cylinder, four stroke engine:
```
1. Detect CRANK TDC. (This could be CYL1 Intake stroke. It could also be CYL1 exhale stroke. Let MAP analysis determine that bellow)
2. During the coming 45 degrees (max) after CRANK TDC, read MAP and wait for drop in MAP.
3. IF no drop in map, stop reading MAP and wait for next CRANK TDC.
4. During the coming 45 degrees (max) read MAP and wait for drop in MAP.
5 .Now Speeduino know that CYL1 is in intake stroke.
```
If reading MAP to get CAM sync for longer than 45 degrees, the risk is that another cylinder starts to inhale and the system is tricked into thinking that its CYL1 intake stroke. Therefore there need to be a tunable window where it looks for sync, depending on the number of cylinders and their crank angle configuration.

Tobbera, 2018-07-13:
After thinking further I realized that this wont work for more than 3 cylinder, 120 degrees crank shaft. That because CYL1 and CYL3 can have intake stroke at the same time after crank TDC.

I have included a picture the the outline of 3 cylinder, 120 degrees crank shaft.  In this case the MAP sensing need to take place between 120 and 180 degrees after crank TDC. This is the **only** time CYL1 alone can be in inhaling stroke. If measuring already from 1 degree after TDC, CYL1 might be mistaken for CYL3 end-of-inhale-stroke, heading for BDC.

[image]

Boogmeister, 2018-07-13:
To add to the above perhaps instead of locking to a certain degree make it calculate current crank angle and use the missing tooth as a reference.

Using this reference decide if within the next x (maybe even a range?) degrees if Cylinder 1 is starting intake stroke or not based MAP?

Tobbera, 2018-08-08:
Done more analysis on this and come to the theory that measuring MAP at pace of once every 10 degree between 50 degrees BTDC and make average och that, called RED. Then measuring between TDC and 50 degrees ATDC and make an average of that, called GREEN. Then comparing RED and GREEN.

If GREEN > RED then CYL1 = TRUE
If RED =< GREEN then CYL1 = FALSE

This would make the method universal for 1-3 cylinder 4 stroke engines. Except for a parallel twin with 360 degree offset crank. I don't think it would work at all on these.

- At idle of about 2000 RPM, the MAP readings would have to be done at a speed of about 1200hz. This leaves a bit more than 1ms for every reading. 

- At 10,000 RPM the MAP readings would have to be done at a speed of about 6000hz. This leaves a bit more than 0.16ms for every reading. 

[image]

Tobbera, 2018-08-08:
Next revolution:

[image]

Tobbera, 2018-08-08:
1 millisecond =1000 microseconds

> "It takes about 100 microseconds (0.0001 s) to read an analog input, so the maximum reading rate is about 10,000 times a second."

https://www.arduino.cc/reference/en/language/functions/analog-io/analogread/

borathtl, 2022-11-30:
Couldn't this be done by just "syncing" the first couple of map readings to every other missing tooth tdcs so the code doesn't have to measure and sample the map at high RPMs? any update on this front?

mattseabrook, 2025-03-23:
Hi everyone — I’ve reviewed the discussions and diagrams (including Tobbera’s excellent analysis) and I’d like to contribute a structured, minimal implementation approach.

---

🧠 **Core Problem Recap:**
We need to identify which stroke (intake vs. exhaust) the engine is on at TDC using only the MAP signal — *without a cam sensor*. This is critical for sequential fuel/spark on multi-cylinder 4-stroke engines.

---

📈 **Why It’s Hard:**
- On engines like 3-cyl 120° crank, two cylinders may have similar MAP behavior within 45–60° windows.
- Measuring too early or too long risks mistaking the intake of CYL3 for CYL1.
- Some engine configs (like 360° parallel twins) produce MAP overlap where this method becomes unreliable.

---

✅ **Solution Approach — Minimal Viable MAP Phase Detection:**

1. **Define windows**:
   - RED: 50° *before* TDC → TDC
   - GREEN: TDC → 50° *after*
   - Sample MAP every 10° (or interpolate based on interrupt timing)
   
2. **Calculate average MAP in both windows**:
   ```c
   float red_sum = 0, green_sum = 0;
   for (int deg = -50; deg < 0; deg += 10) red_sum += map_readings[deg];
   for (int deg = 0; deg <= 50; deg += 10) green_sum += map_readings[deg];
   float red_avg = red_sum / 5;
   float green_avg = green_sum / 6;

3. Decision Logic:
    ```c
    if (green_avg < red_avg - threshold)
        assume intake stroke (MAP drop detected)
    else
        wait for next cycle
  
- threshold can be calibrated (e.g., ~2–4 kPa drop) based on engine behavior.
- Lock phasing once a clean signal is seen — this removes ambiguity for subsequent injector/spark firing.
- At 10,000 RPM, we get ~333 µs per degree.
- Sampling every 10° gives ~3.3ms total window — easily achievable, even with basic analogRead (~100 µs per sample).
- Sampling MAP on crank ISR tick with interpolation is viable on Atmega2560-based platforms.

The attached .png and .jpg files in the comments above — those reinforce the theory that a MAP-based inference window only exists between 60–180° ATDC for 3-cylinder, 120° crank engines. That means we must be ensured no overlapping inhaling cylinder during the read window.