speeduino/speeduino #146: Duty cycle stops att 24.5

KylTony, 2018-06-24:
4squirts alternating basictrigger v8 2injectors.
Duty stops att 24.5% ????? and it goes lean.
Max duty is set to 98%
Swapt to MS2 and it goes upp to 95% att full throttle.

noisymime, 2018-06-25:
Please post tune and log files of this issue occurring.

KylTony, 2018-06-25:
[Duty.zip](https://github.com/noisymime/speeduino/files/2131954/Duty.zip)

KylTony, 2018-06-25:
[Duty.zip](https://github.com/noisymime/speeduino/files/2131960/Duty.zip)

KylTony, 2018-06-25:
ops bigg fingers

lsdlsd88, 2018-06-25:
could it be due to 4 squirt per cycle screws up some DC% calculation? 24.5x4 is 98...

noisymime, 2018-06-25:
@lsdlsd88 This was my first thought as well. I'll be looking at this type of thing shortly.

kimdrummel, 2018-07-18:
I can confirm this problem as well. Tried it out today. When selecting 1 squirt/cycle, you get the full duty. But when selecting 2 squirts, you get 50%, 4 squirts 25% etc. Using Tony's tune.

lsdlsd88, 2018-07-18:
Yes but does 50% for 2 squirts actually means saturating injectors?
I mean if the issue is only in representing real DC% or it effectively cuts the squirt duration in half (you'll get a very lean condition)

KylTony, 2018-07-18:
[image]

kimdrummel, 2018-07-22:
Tested this on the bench with older firmwares. And up until the February release, everything works fine. But when loading the April release, the duty and pulsewidth is starting to get limited. On the February release, i get a full 100% duty cycle and about 7ms opening time max. (Req fuel is around 8) But when running the April Firmware, it gets limited to 25%, followed by a limiting of the pulsewidth as well.

noisymime, 2018-10-02:
Just following up on this, the issue should've been resolved a little while back. Are you able to confirm whether this is resolved with the current release?

KylTony, 2018-10-02:
Hey! I have tested aug firmware and now 2 squirts work simultaneously / alternatively.
4 squirts still not work stops at 23.7 in duty cycle 😢😭

noisymime, 2021-02-22:
Closing as I cannot reproduce on any recent firmware. Please reopen if this is still a problem on master code.

closed 2021-02-22 by noisymime