speeduino/speeduino #151: Teensy fails to boot when activating Thermo Fan function.

kimdrummel, 2018-07-18:
I have shed some light over this issue in the Forum, but thought it might be a good idea to add it here too.

When activating the Thermo Fan function, the teensy won't reboot properly when restarting the system. You need to do an eeprom clear and then load the firmware again to make it boot.

It's like it won't read the program properly and gets an error or something in the booting process when the Thermo Fan is activated.

I have tried this with 2 different teensys. And it also happens no matter what MSQ you load.

noisymime, 2018-07-18:
In your testing, is this only happening when also connected to 12v rather than just USB?

In my testing, I'm still seeing the unit boot fine with the fan enabled, though I haven't tried it with 12v yet

NickZ1969, 2018-07-29:
ive been running the thermo fan on (enabled) and haven't had any problem with my teensy 8x8 ecu.
maybe you have a conflicting pin?

0x15e, 2018-08-23:
This sounds a _lot_ like what I experienced during my brief experimentation with using a Teensy. Mine is always powered via 12V. I assumed my Teensy or early revision adapter board was the culprit and went back to Mega.

I do see that #156 was supposed to address an issue with Teensy on the 0.4 boards. Any chance that may have helped this? @kimdrummel what revision speedy are you using?

kimdrummel, 2018-08-23:
I use my own board. But it basically is a 0.4 with my own arrangement and everything SMD with the Teensy and Teensy Adapter components put directly on the main PCB.

noisymime, 2022-10-03:
Assuming this was still an issue, it would have been resolved with a4b00dca9011da6b297655ba41182f25940de700

closed 2022-10-03 by noisymime