speeduino/speeduino #153: Questionable compare logic

TradieUte, 2018-07-24:
April rls code. 
Speeduino.ino ln 248. 
configPage2.pinMapping - which board I'd to use; can be > 100.
BOARD_NR_GPIO_PINS - the number of pins defined for board;  defined for Teensy as 34. ( should this be 57 for Teensy 3.6?)
The compare gives incorrect results - loads wrong board pinout -  when board I'd larger than 34. My project using board I'd = 60.
Results in several variables being used without being initialized. Doesn't stop code running, but delivers unexpected results - secondary serial stops working.
Need a compare value that is dimensionally the same as configPage2.pinMapping value.

noisymime, 2018-07-24:
This is a fair point. Hasn't been an issue as no one is using board IDs that high to date. I'll commit something shortly to resolve this

closed 2018-07-24 by noisymime

TradieUte, 2018-07-24:
Thanks Josh,
No hurry. Now that I have identified the cause of the weird problem I had, I can work around it.
RegardsAndy 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/noisymime/speeduino/issues/153#issuecomment-407287905",
"url": "https://github.com/noisymime/speeduino/issues/153#issuecomment-407287905",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [noisymime/speeduino] Questionable compare logic (#153)",
"sections": [
{
"text": "",
"activityTitle": "**Josh Stewart**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@noisymime",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"noisymime/speeduino\",\n\"issueId\": 153,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"noisymime/speeduino\",\n\"issueId\": 153\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/noisymime/speeduino/issues/153#issuecomment-407287905"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 359687557\n}"
}
],
"themeColor": "26292E"
}
]

commit 45bbb4ba (2018-07-24): Correct check for new board on startup. Fixes #153