speeduino/speeduino #154: Water / Meth injection

DeeEmm, 2018-08-11:
This is a feature request / firmware enhancement.

I would like to see the integration of water / meth injection control, both a basic on/off version triggered at a pre-determined boost level and also a ratio-metric / dynamic system using MAP or TPS and RPM to enable, a simple offset table with +/-100% setting based on the current injector pulse width along with a 2D ignition correction table.

Some basic discussions relating this this can be found here - [
https://speeduino.com/forum/viewtopic.php?p=27696&sid=b8bb3b67641433a405b017719837a036#p27696](url)

Safeties / failsafes to be considered and at a minimum to include water level sensor to cancel ignition table correction (user enabled).

AU$200 bounty offered

DeeEmm, 2018-08-11:
Oops, think I raised it in US$ instead of AU$. 

Maybe we could include a shift light along with it too ;)

ByteVenom, 2018-12-09:
Are you looking for this to run off the speeduino itself? I think this might be more of a GPIO solution.

DeeEmm, 2020-06-17:
Bumping this as the bounty is about to expire and will be lost :(

DeeEmm, 2020-06-17:
Deadline is 11 August

DeeEmm, 2020-06-17:
Link to discussion on Speeduino site above seems to be dead. You can access it at the link below:

[https://speeduino.com/forum/viewtopic.php?f=13&t=2358&p=43710#p43710]

DeeEmm, 2020-07-22:
Bounty solved by Timu5 PR#417

Great work. Thank you !!!

closed 2020-07-22 by DeeEmm