speeduino/speeduino #161: AFR Safety System

noisymime, 2018-09-05:
Per forum: https://speeduino.com/forum/viewtopic.php?f=12&t=2048

Feature request:
A simple form of afr safety, so you dont blow your engine if a pump or an injector fails. 
Setup would be something like:
1: Above X kpa
AND/OR
2: Above X tps
IF
3:Afr not below X
IN
4: X seconds
DO
Cycle some ignition cuts so the driver feels somethings wrong AND/OR turn of boost control/lower boost

zrparker, 2018-09-19:
Wouldn't it be better and maybe easier to have something like:
IF AFR>(TARGET AFR+SET VALUE)
 Cut spark

That way it will be checking at all speeds and loads. Or, heck, tossing in a MAP or TPS min would make it so that it couldn't cut under low loads.

ric355, 2019-11-21:
Must account for DFCO bit being on, where AFR will shoot up very lean. Probably wouldn't make a difference immediately as there is no fuel at that point, but the time period between fuel coming back on and AFRs responding is relevant.  Note also that checks against AFR must account for the time delay which is both load and RPM based and different for every engine.

ric355, 2019-11-21:
Must also not interfere with the lean spot you tend to get when accel enrichment isn't tuned well enough.

zrparker, 2019-11-23:
> Must account for DFCO bit being on, where AFR will shoot up very lean. Probably wouldn't make a difference immediately as there is no fuel at that point, but the time period between fuel coming back [...]

TPS min would account for DFCO. Time delay is already accounted for in the original idea and so is the accel enrich issue with this:

>IN
4: X seconds
DO

ric355, 2019-11-23:
> TPS min would account for DFCO. Time delay is already accounted for in the original idea and so is the accel enrich issue with this:

I think the original time delay is intended to describe that the AFR would need to be outside of limits for X seconds for the warning condition to be met. That's sensible but it is not what I am referring to. I am referring to the fact that when you read the O2 value there is a time delay that must be accounted for when doing comparisons between some set value and the "current" value in relation to current fuelling, because the current sensor value is from an ignition event that happened at an earlier time. This maybe doesn't matter for this specific check, but the reason I mentioned it is that I know that the lambda delay table is not currently used by Speeduino (I mean the one used for AutoTune) as it relies either on instantaneous readings or on PID control.

The DFCO issue is not fully accounted for by TPS min. You can be completely off throttle for many seconds (e.g. going down a steep hill) and without a check for DFCO active you would need either a very very long timeout (which makes the change pointless) or would get a lot of false triggers.  It's not a big deal, it's just a simple check that DFCO is not active as part of the 'and' condition.

zrparker, 2019-11-27:
> I think the original time delay is intended to describe that the AFR would need to be outside of limits for X seconds for the warning condition to be met. That's sensible but it is not what I am [...]

That's exactly what the time delay portion is for.

Lets be honest, the odds of joe shmo getting that time delay correct are nearly zero because there are too many factors at play that can't be measured/accounted for with the I/O that is generally hooked up with these. Yes, you could theoretically get a bit closer, but that's about it. I'm not a fan of AT in any way because I don't like the control you are giving up, but that's a whole other discussion.

> The DFCO issue is not fully accounted for by TPS min. You can be completely off throttle for many seconds (e.g. going down a steep hill) and without a check for DFCO active you would need either a [...]

You don't really need it to be. If you are in DFCO, you are, by definition, not under load. Because you aren't under load, there is no point in the AFR safety. In other words, no TPS means no load which means no AFR safety required.