speeduino/speeduino #164: Accel Enrichment tapering appears to be inverted

ric355, 2018-09-10:
The June release seems to have the accel enrichment taper operating the wrong way. It adds zero percent of the accel fuel at the low RPM point in the taper and 100% accel fuel at the high RPM point.  This results in a sudden drop in accel enrichment after you cross the taper point on the way up the rev range. I believe this is the opposite to how it should work; accel enrichment should be less at high RPM and more at low RPM.  Here's the fragment from corrections.ino; 

`        accelValue = table2D_getValue(&taeTable, currentStatus.tpsDOT);

        //Apply the taper to the above
        //The RPM settings are stored divided by 100:
        uint16_t trueTaperMin = configPage2.taeTaperMin * 100;
        uint16_t trueTaperMax = configPage2.taeTaperMax * 100;
        if (currentStatus.RPM > trueTaperMin)
        {
          if(currentStatus.RPM > trueTaperMax) { accelValue = 0; } //RPM is beyond taper max limit, so accel enrich is turned off
          else 
          {
            int16_t taperRange = trueTaperMax - trueTaperMin;
            int16_t taperPercent = ((currentStatus.RPM - trueTaperMin) * 100) / taperRange; //The percentage of the way through the RPM taper range
            accelValue = percentage(taperPercent, accelValue); //Calculate the above percentage of the calculated accel amount. 
          }
        }
        accelValue = 100 + accelValue; //Add the 100 normalisation to the calculated amount`

The current code calculates what percentage of the way through the taper range we are, and uses that directly as the percentage of fuel from the table lookup. This means bottom of the range = low fuel amount, and top of range = high fuel amount, which according to everything I've read is the opposite to how it should be. We could change to:
`accelValue = percentage(100-taperPercent, accelValue); //Calculate the above percentage of the calculated accel amount.`

I believe this will invert the result to the proper amount.

Alternatively I think you could change to this, but I've only tested the first suggestion:
`            int16_t taperPercent = ((trueTaperMax - currentStatus.RPM) * 100) / taperRange; //The percentage of the way through the RPM taper range
`

noisymime, 2018-09-10:
Crap, I know exactly how this happened in my testing too (Was testing reverse TPS stuff at the same time).

Thanks for the report, I'll get a fix out shortly.

noisymime, 2018-10-02:
Confirming that this is working now following 5d271ef

closed 2018-10-02 by noisymime