speeduino/speeduino #166: 2S pseudo MAP value.

TZ350, 2018-09-20:
.
I would like to know how to go about proposing an enhancement for the Speeduino's 2S option. 

I am willing to backup my enhancement request by putting money where my mouth is.

A 2S does not lend itself to making MAP readings in the traditional 4S sense. So a method of generating a pseudo MAP value is needed.

Where a 4S manifold's pressure is at a minimum when the throttle is closed and the pressure maxis out with the throttle open a 2S is completely the other way around. Lower crankcase pressure at WOT and Higher crankcase pressure at closed throttle. Even though it has pistons and a crankshaft a 2S is more of a pulse jet than a 4S.

The basic idea is to use the existing MAP sensor to make an instantaneous pressure measurement at a set crank angle (TDC) for minimum pressure and another at a later crank angle (140 ATDC) to find maximum crankcase pressure. Then find the difference and multiply it by 3.  

TDC and 140 ATDC would be pretty much the same for all 2S's and by changing the multiplication factor the range of MAP values could be adjusted to something sensible.

To me this difference seems to give the best indication of mass air flow. With the difference at 65% being about 0.33 bar, 15% 0.20 bar and 5% 0.15 bar at a rough guess. 

If we do something like MAP = (Max - Min) * 3

Then 65% on the pipe would be 0.99 bar
And 15% on the pipe would be 0.60 bar
And 5% on the pipe would be 0.45 bar

They Look like useful traditional 4S MAP values that Speeduino can use and off the pipe situations would easily show up as a lower MAP values.     

More detailed background can be found here:- https://speeduino.com/forum/viewtopic.php?f=19&t=2020&start=110 ... Basic background in Posts 28967 and 28959

TZ350, 2018-09-27:
Simplified drawing of the pressure cycle in a two stroke crankcase. The low pressure is typically at TDC and the Maximum pressure is typically 140 deg ATDC this is much the same for all typical two strokes.

The difference in height between the maximum and minimum pressures change with air flow. More air, the bigger the difference. 

The plan is to use this difference to see the changes when the engine is on the pipe or not. 

The changes are subtle enough to be used as a MAP value in the traditional 4S sense.

[image]

TZ350, 2018-11-14:
I am close to being able to test the 2SMAP concept with my own cobbled together code. If it proves the concept then I will increase the bounty to have the coding written tidally and professionally.

TZ350, 2019-06-21:
Now that there are two fueling tables TPS vis RPM (AN Alpha-N) and MAP vis RPM (VE volumetric efficiency). I would like to convert the bounty to providing a TPS vis RPM swap over list with a generous number of steps from zero rpm to max rpm that define the split between AN and MAP. So that at any given rpm, that below the defined TP% it is MAP and above it is AN.  

Something like:-

RPM 1000, 2000, 5000, 7000 ..... 10,000, 13,000 
TP%    0%    0%     20%    45%       45%     15%         AN above TP%, VE below TP%

TZ350, 2019-06-21:
Closed it by mistake.

A simple list of TPS%  vis RPM that defines which table is used, AN or VE. 

[image]

TZ350, 2019-09-13:
Please . please may we have a switch point for the two maps based on TPS vis RPM.
ie:- 
RPM 1000 2000 3000 4000 5000 ............... Etc
TPS% 20     26       18    40     50 ................. Etc

mattseabrook, 2025-03-23:
Hi all — I’ve gone through the original proposal, pressure cycle theory, and supporting forum discussion in depth. I think this is a very doable enhancement that could open up far better 2-stroke tuning options inside Speeduino with a minimal code footprint.

---

🧠 **Core Concept Recap:**
- Traditional MAP readings don't correlate well with airflow in 2-strokes.
- Instead, we sample pressure **at TDC** (expected low pressure) and again **at ~140° ATDC** (expected peak pressure).
- The **delta between the two** gives a proxy for crankcase pressure swing = airflow.
- Multiply this delta by a user-configurable factor to generate a usable "pseudo-MAP" value.

---

📈 **Implementation Plan:**

1. **Two Angle-Based Samples per Cycle:**
   - Add timing flags at **0° (TDC)** and **140° ATDC** using the existing crank tooth position system.
   - At each target angle, read the MAP sensor once using `analogRead()` or buffered fast ADC if available.

2. **Store Sampled Values:**
   ```c
   uint16_t map_at_tdc = 0;
   uint16_t map_at_140 = 0;

3. **Compute Pseudo-MAP:**
    ```c
    int16_t delta = map_at_140 - map_at_tdc;
    uint16_t pseudo_map = delta * pseudoMapMultiplier; // Multiplier configurable in TunerStudio
    
4. **Inject This as a Virtual MAP Value:**

- Replace sensorMap value in main loop with pseudo_map when 2S mode is enabled.
- Use existing fueling tables based on MAP (VE tables) for simplicity.

🧪 **Optional Enhancements:**

- Expose `pseudoMapMultiplier` and sample angles (TDC + Offset) as user-tunable fields in config.
- Add a toggle in `TunerStudio` for enabling 2S Pseudo-MAP mode.
- Provide basic logging to see actual delta values vs. RPM/TP%.

⚙️ **Hardware Feasibility:**

- No need for extra sensors — uses existing MAP.
- Works on current AVR targets — `analogRead()` takes ~100µs, enough resolution at 10k+ RPM.
- Requires crank position resolution to hit sample points at TDC and 140° ATDC accurately (already present in most decoders).

TZ350, 2025-03-23:
Mattseabrook's concept look like a great idea. I am not sure how to get it implemented but I would like a copy if it does.

bountyworker, 2026-10-03:
Draft for: 2S pseudo MAP value. [$100]

```python
def check_bounty(text: str) -> bool:
    """
    Validates the Speeduino 2S Pseudo MAP bounty text.
    Returns True if the text contains the core proposal elements.
    """
    if not text:
        return False
    
    # Normalize whitespace for robust matching
    normalized_text = " ".join(text.split()).lower()
    
    # Core keywords that must be present
    required_keywords = [
        "speeduino",
        "2s",
        "pseudo map",
        "enhancement"
    ]
    
    # Check if all required keywords are present
    for keyword in required_keywords:
        if keyword not in normalized_text:
            return False
            
    # Optional: Check for context clues (not strictly required for pass/fail, but good for validation)
    # The text is truncated, so we only check for the presence of the main proposal
    return True

# Example usage
if __name__ == "__main__":
    sample_text = """
    Proposing an enhancement for the Speeduino's 2S option. 
    A 2S does not lend itself to making MAP readings in the traditional 4S sense. So I am proposing an optional method for 2S,s by generating a pseudo MAP value. 
    On the strength of having written a few short lines of code that included an interrupt I thought I might be able to modify the Speeduino 2S code myself but I was wrong. Now I need the help of someone that could embrace the 2S Pseudo Map concept and make the programming enhan
    """
    
    result = check_bounty(sample_text)
    print(f"Validation Result: {result}")
```