speeduino/speeduino #168: Trigger Jumping around 24/1 toyota trigger. [$50]

Josh4age, 2018-10-09:
Dear Gentlemen,

I have been struggling with the trigger pattern for dual wheel cam 24 tooth primary wheel (12 at crank speed) and 1 tooth wheel also at cam speed. The car is a toyota ke70 with a 4age 1.6 smallport 16v engine swap. Running the stock distributor and single coil + ignitor setup. paired injectors on the two channels.

I am using the no2c board running June firmware, Vr conditioner max9266 Josh stewart special. 

Basically the problem is that every time the car starts up the ignition timing will be at either one of three points, I have set the trigger offset to -15deg because it will put all three points at a spot where the car can run. The values will be at ignition locked to 15deg

1.  40-50 deg BTDC
2. 15 deg BTDC (bang on the locked timing
3. 10 deg ATDC

Now the car has 3 VR sensor pickups 1 on the 24 tooth wheel (NE) and 2 on the single tooth wheel 180 Deg apart (G2 at TDC CYL1) and (G1 at 180deg offset). Initially I started with NE as primary wheel on 12 teeth at crank speed. and G2 as VR2 leaving G1 unused, grounding it at the ecu spare grounds. I couldnt get the car to run right until used medium trigger filtering and no for resync at every cycle. I thought the problem was fixed but a few days later it seems that the problem was still there.

Next I thought that maybe it was the unused VR sensor causing problems so i removed it. also all three sensors share the same ground so I decided the separate the grounds by grounding G2 sensor using the G1+ wire, so basically NE and G2 have their own separate grounds and the unused sensor is our of the distributor.

I am still getting the same problem, I have also played around with the trigger filtering and also resync but nothing really seems to change. 

Please let me know what I should try next, if there is anything you need please let me know I can get access to an oscilloscope, or buy tunerstudio to get the tooth logs. 

Thanks for your time and sorry to be a bother. I would like to put a bounty on for AUD50 if that will help

[image]

[image]

noisymime, 2018-10-09:
When you say grounding on the sensors, do you mean the shield on the sensor wires or are you actually running one of the VR signal pins to ground?

Each VR sensor should have 2 wires (and optionally a 3rd one for the wire shield). those 2 signal wires need to be connected to VRx+ and VRx- on the board (Where x is either 1 or 2 for primary or secondary). You shouldn't be running either of those signal wires straight to ground.

Josh4age, 2018-10-09:
Hi Josh, Thank you for the quick response. I dont see any shielding with the cables, Im sure they must have grounded the shield in the loom somewhere. Yes i was grounding the signal wire to stop it from floating, but thats all gone now, the current setup is 2 sensors with their pairs of wires vr+ and Vr-. I thought this might help, but it didnt make any changes, I have cranking revolutions before start up set at 2 rev, tried 1 and 0 almost flooded a couple times. 
Could it have anything to do with the shape of the single tooth wheel? its like a increasing gradient to a point. 

I have a slight feeling that when it syncs at the beginning it doesnt always sync correctly? depending on where the engine is before you start determines which trigger you are going to get, because there has been many instances where while driving, the revs drop below say 500rpm it will just die and then start up all funny at a different sync angle either retarded or way advanced. 

I hope you can make some sense of all this.

Josh4age, 2018-10-11:
OK today tried to run a Shielded cable from the distributor to the speedy. The v. 04 board has some problems it longer picks up rpm, this happened yesterday after cranking. Also I noticed that the led for injectors 1 stayed on. So something is wrong with  that board.

I will go on testing using the no2c which still functions well.

The Shielded cable is really small they are about 24awg gauge which is a whole lot thinner than factory. Also it is about 6m long I didn't shorten it. Earthed the shield on one of the ecu grounding pins.

Still getting the same problem, I changed up the filter settings without much luck. I put it on resync after every cycle and it's even worse it changes timing while revving the car. I didn't get time to try aggressive. After that I may have exhausted all my options on the car side of the issue. 

The last thing I can do now is try a multipoint grounding kit that hks do. Hopefully that will help out with all the sensors.

Josh4age, 2019-02-24:
Problem unsolved tried following

1. dsc vr conditioner board, runs but problem is still there.
2. LM1815 board for processing vr2 signal - didnt work because i believe the voltage is too low on the single tooth trigger
3. LM1815 with a variable resistance pot, starts easier but still dies after a few seconds, at varying resistance, doesnt filter out the noise causing the problem. 
4. Thought about using IC hall sensors like the ones that Isaac dalton suggested however i just dont want to place magnets in there what a pain in the neck!

The only feasible solution i have found is to write a crosstalk vr signal filter that will rub out the noise signal causing the confusion to the speedy as to which tooth came after the home signal. there is a software vr filter in the MS3 code, If someone could port it over to speedy code the problem will be fixed running a DSC or MAX vr conditioner chip.

So until then I am going to give up. this is where my capabilities are exhausted.

kijananick, 2020-11-02:
Hello josh...did u manage to get a solution...I seem to have run into the same problem

Josh4age, 2020-11-02:
Hi Josh,

Thanks for your reply, grounding is the term used by nippon denso, as both
vr sensors have + and - outputs, however the (-) output of all three
sensors are bound together and are referred to common ground for the
sensors designated as G-. This presents an opportunity for crosstalk to
take place. Nippon densos hardcore circuitry is able to filter out these
cross talks without having to shield the vr wires. Speeduino will need some
software filtering to filter out some of these signals, because it seems to
be confused as whether the sync tooth is coming as the 11th, 12th or 13th
tooth, which explains why it is out in increments of 15 degrees.

Josh

kijananick, 2020-11-02:
Thanks for the response josh...so in simple terms u just added software filtering and the timing drift went away...I have tried locking timing at 0 degrees to set the trigger angle...but the timing always drifts away when I am just about to set it at 0 and the car stalls.when I restart the car...the timing is at a different location

noisymime, 2020-11-02:
I've had a chance to play with these Toyota setups a little more since this issue was originally raised and have found that the best option (And by best I find it to generally be the simplest whilst providing a very good result) is to run a couple of extra wires so that each sensor can have it's own negative. 

I've played around with different conditioner options, different software filtering techniques and even swapping the sensors for gear tooth ones, but ultimately found that spending an hour or 2 just running and joining in the extra wires works really well every time. It means you can use a regular MAX9926 based conditioner, all the normal software filtering etc, and the signal is rock solid. 

I know it's not necessarily the answer people want, but I can say that it works very well.

Josh4age, 2020-11-03:
Not quite, Ive switched to a Microsquirt ECU which has all VR conditioning circuits built in as well as advanced adaptive vr filtering options. Since then I haven’t had any problems (apart from breaking diffs and gearboxes) anyways if ur interested in one I have a spare one set up for 16v smallport or 4agze loom plugs. you can catch me on facebook under Josh Ng

victorestebanr, 2021-01-12:
Greetings all,
I will start wiring my Speeduino UA4C next week and know will have the same type of problem.
I have a 4EFTE engine out of a 1998 Toyota Starlet Glanza, this engine uses the 24/1 crank/cam wheel setup. I would like to keep it as stock as possible but run COP with wasted spark. I've read several forums, some indicate that I can get it working, others say that I will either have to grind one of the 24 tooth on the wheel or run a crank mounted 36-1 one with an external sensor. 

While the external crank wheel seems easy I don't really want to mess with that and want to retain the PnP capabilities in case I want to return to the OEM ECU.

I will attach some pictures that I hope help,

Appreciate your help. 
[image]
[image]

https://user-images.githubusercontent.com/77362402/104378784-eaa46200-54fe-11eb-9419-b023b1267ae1.mp4

[image]
[image]
[image]
[image]
[image]
[image]

dascons, 2021-01-13:
I believe this is for firmware and hardware issues/ feature requests etc
and not for vehicle help. Use the speedy forum.
That said, i'd recommend not using the factory wiring going to the sensor.
Inside of the unit, the "grounds" of each sensor are connected together.
The VR conditioner that we use does not like this.
I would recommend running your own wires to the CAS, with two separate
pairs so that the sensors are separated from each other.
A crank mounted sensor would be best but it really is more of a pain than
necessary (unless you plan on cams as this makes the sensor reading jump
around a bit too much for my liking)