speeduino/speeduino #169: Strange basic distributor behavior

jmunkdk, 2018-10-18:
Hello,

I have for as long as I have attempted to get the Speeduino working on my 1975 VW Bus had issues starting it. Now that the weather is colder it is basically a no go, so I decided to check the starting attempt with a logic analyzer. There is certainly something wrong, and both ignition and injection seems rather random.

The trigger is a hall sensor in a basic distributor (with advance locked). The distributor is an original electronic Bosch from a slightly newer VW, but is working perfectly.

[image]

[VW start 01.zip](https://github.com/noisymime/speeduino/files/2493334/VW.start.01.zip)
(PulseView from http://sigrok.org. Can provide other formats)

As you can see it is firing and injecting all over, which is rather odd since "Fix cranking with trigger" is on. Only slightly special observation is that the cranking speed is not very high for these old VW engines, but this is normal. They are also normally very easy starters, and this is probably why I have been able to start it at all. Anyway, all other details should be in the tune file, or ask.

[CurrentTune.zip](https://github.com/noisymime/speeduino/files/2493307/CurrentTune.zip)

Besides reporting this issue, I wonder if a 36-1 missing tooth trigger like this would resolve the issue:

http://thedubshop.com/type-4/

Thanks, 

Jens.

PSIG-dev, 2018-10-19:
I assume you are using the latest AUG18-2 firmware and INI?  There may be a code issue, but your trigger signal is rather messed-up.  Not just slow.  If you can fix that, it would make diagnosing any ECM issues easier.  Please post a short log of engine starting using your current tune.  

If you tested the trigger (tach) signal at Speeduino input, please test again at the D19 pin on the MEGA, or other nearby place on the board, such as IC3 pin 8 hole.  Compare with a test directly at the sensor connector.  This will narrow the source of the weird distributor signal.  Are you using any type of signal conditioning circuit?  This is not possible if working correctly:
[image]
Perhaps posting all of this in your forum project thread would be more effective.

jmunkdk, 2018-10-19:
Requested new measurements are ready. The zip file contains the tune, and a set of a MT log and a logic analyzer log. Just to be absolutely sure there is no noise sources, I have disconnected injectors, ignition coil, ICV and fuel pump. The trigger is measured at IC3 pin 8, and it is clean. The sample rate is 16 MHz, so nothing should be missed. The pulses match the RPM in the TS log of about 134 rpm on average, so it must read them correctly. Ignition and injection only seems sporadically correlated to the trigger.

[image]

[WV Start 02.zip](https://github.com/noisymime/speeduino/files/2496101/WV.Start.02.zip)

What do you think?

PSIG-dev, 2018-10-20:
Yes, your second LA image shows a much cleaner and more consistent trigger pulse, with a different system response.  This will make some diagnoses by @noisymime much easier, but also means you have some signal cleaning to do (likely identifying EMI to fix or avoid, based on the signal changes). 

The "after-run" pulses are _generally_ not an issue, as either or both Speeduino and the components (ignition, injectors, etc) are not powered or power is removed when keyed-off.  Speeduino is attempting to predict proper spark and fuel with last-known information, while still powered and functional.  In other situations (e.g., engine stall), the RPM reduces (even if very quickly) and after-run pulses are not an issue _that I have seen_.  If you have a situation where trigger pulses stop instantly, but the Speeduino and the components remain powered, and after-run pulses continue in a potentially damaging way; please describe it so that appropriate solutions may be pursued.

jmunkdk, 2018-10-21:
Thanks,

After my last post, I did some more measurements, and most notably it appeared that the trigger filter was set to OFF. This is the default setting in speeduino.ini, so I suggest you change that default to Medium.

Setting the trigger filter to medium seems to get fairly solid ignition and injection pulses, but it now takes between a couple of trigger pulses to 5 - 10 trigger pulses before it starts to ignite and inject. One time it didn't start pulsing at all. Skip revolution cycles (1 and 0 tried) doesn't seem to make any particular difference and changing the trigger angle between -5, 0 and 180 neither. 

The description of "Fix cranking timing with trigger" says "On certain low resolution ignition patterns, the cranking timing can be locked to occur when a pulse is recieved.", but it doesn't seem to make any difference either.

The New Ignition mode left one of the injectors permanently on when I stopped cranking!

I see your point in the after cranking pulsing not being relevant, since you turn it all off. Well I then certainly need to get accustomed to turn the the key all the way to off when it doesn't start. Not too crazy about this, but OK, I can live with it for now.

So I think some more optimization is required for the simple distributor mode.

Having said all that, how will a 36-1 hall trigger work in this respect? Any trigger guessing? Any after crank pulsing?

Thanks,

PSIG-dev, 2018-10-22:
Good suggestion — we can ask for outputs to be clipped/cut after a limited or configurable time in cranking mode.

The trigger filter is default to "Off", assuming a clear and clean signal.  Which setting to use if yours is not is then up to you, as (like you've seen) it carries compromises, and each situation is different.

@noisymime should perhaps have a look at "Fix cranking timing with trigger", as it should clearly be _**if A then B**_, where an input signal directly results in an output.  There are others using Basic Distributor in this way without claimed issues, so I'm not sure of the differences you are seeing.

Low-resolution timing control is extremely challenging.  Consider an analogy, where you are playing ball with someone, and you know they are going to throw the ball sometime, but you only get to look for a moment every few seconds.  Good luck... LOL  A 36-1 would have greater resolution and fewer potential issues trying to time spark outputs.  There is little 'guessing' for setup.  'After-pulsing' would be the same as currently.

TBAMax, 2021-10-12:
Just a note here for now. Recently I discovered that there is a problem with  "per tooth ignition"  running with "fixed cranking ignition override". Should be one or another but not both.

noisymime, 2024-03-14:
Closing this as stale. I believe this has likely been addressed in other changes that have gone in for this decoder over the past few years. 

If this remains a problem, please create a new issue for this.

closed 2024-03-14 by noisymime