speeduino/speeduino #170: Limited Injector Duty Cycle: 30% on 6 cylinder, 3 squirts per cycle

V6Vintage, 2018-10-22:
Rather than leave random comments in Slack i thought i'd post here. 
6 cylinder: anything more than 2 squirts per cycle gives limited injector duty cycle.

Duty cycle also 'spikes' to the max number, 30 in my case.
If set to 6 squirts per cycle, max duty cycle is limited to 15.

Attached my current tune and a short log file *you'll have to rename them as i can't make zip files on my linux laptop!* sorry

Attached 
[V6Vintage_log_renametoMSL.txt](https://github.com/noisymime/speeduino/files/2502551/V6Vintage_log_renametoMSL.txt)
[CurrentTune_renametoMSQ.txt](https://github.com/noisymime/speeduino/files/2502554/CurrentTune_renametoMSQ.txt)

noisymime, 2018-10-22:
Confirmed there is definitely some incorrect behaviour with 3 squirts. There's some bad logic at one point in the setup code, specifically for 3 squirts, that needs resolving. 

It's not entirely clear how 3 squirts should best be handled at this point as it's only a valid over pattern over 720 degrees (as in the pattern is not 'symmetrical between 360 and 720 degrees), which technically requires a cam reference to identify the phase. I'll have a play around here, but for the moment this is a valid bug for 3 squirts.

V6Vintage, 2019-05-13:
I have (hopefully) posted a bounty on this issue as i'd really like 3 squirts per cycle! I have one injector per 3 cylinders, so 2 squirts per cycle doesn't distribute nicely.

I have tried to solve the issue myself but i'm missing something obvious. It's above my head!

noisymime, 2019-05-13:
Hmmm, I had actually thought this was resolved some time back. I'll have another look at this shortly and see what's going on.

The point above about potentially needing a cam signal still stands though. Because 3 squirts must be distributed across 720 crank degrees, it requires a trigger arrangement that runs over 2 engine rotations.

V6Vintage, 2019-05-14:
What about for 6 squirts alternating? wouldn't that give three injections per side, per cycle? (just to recap my V6 engine has an intake manifold with two halves and two injectors. so one injector per three cylinders). at 90% max allowed duty cycle for the injectors, the highest pulse width i can get is 15%. Thanks for your help! why hasn't my bounty appeared yet?! it's $50 anyway.

V6Vintage, 2019-05-14:
in speeduino.h, line 402-403 there is:
`if (CRANK_ANGLE_MAX_INJ == 720) { pwLimit = pwLimit * 2; } //For sequential, the maximum pulse time is double (2 revolutions). Wouldn't work for 2 stroke...
      else if (CRANK_ANGLE_MAX_INJ < 360) { pwLimit = pwLimit / currentStatus.nSquirts; } //Handle cases where there are multiple squirts per rev`

What happens if CRANK_ANGLE_MAX_INJ  = 360?  should it be:
`if (CRANK_ANGLE_MAX_INJ == 720) { pwLimit = pwLimit * 2; } //For sequential, the maximum pulse time is double (2 revolutions). Wouldn't work for 2 stroke...
      else if (CRANK_ANGLE_MAX_INJ <= 360) { pwLimit = pwLimit / currentStatus.nSquirts; } //Handle cases where there are multiple squirts per rev`  ?

I may try this tomorrow where i have the day off - it's currently nearly half past 9 pm and the neighbours won't like me backfiring on my driveway :D

richcreations, 2019-05-15:
I thought the “fix” was to set it up as if you had batch injectors (3channels) and combine them with diodes, at least that’s what I did.

V6Vintage, 2019-05-15:
Which bit did you set as batch, the software settings? Because the outputs behave correctly as batch without any diodes...   Or do you mean set the software as 6 squirts alternating and physically combine them with diodes so they act as batch?  In that case I'd imagine low rpm tuning would be tricky as the open times would be pretty small. The issue at hand is removing the glass ceiling of PwLimit at higher load and RPM. thanks

V6Vintage, 2019-05-15:
so: an absolute bodge here but it works for my setup:
speeduino.h, line 377:
`else if (CRANK_ANGLE_MAX_INJ < 360) { pwLimit = 90000;}  //Handle cases where there are multiple squirts per rev`

I've hardcoded the pwLimit to 90,000.  60,000 makes it top out at about 60% in real life so i assume 90,000 will be 90% - i can now accelerate as hard as i like and everything feels completely normal.
My engine is running 6 cyl, 6 squirts alternating, Throttle body.

My earlier 'hack' of setting the engine up as 3 cyl didn't really feel right. I definitely have all cylinders getting fuel now...!

noisymime, 2019-05-16:
OK, this was a relatively complex issue, but I'm 95% sure it's resolved with that last commit. 

Note that now for running 3 or 5 squirts, you MUST have a trigger arrangement that allows for sequential (ie you have a cam signal). You don't actually have to run sequential, but the system must be able to track crank angle over 720 degrees.

noisymime, 2019-05-16:
@V6Vintage Setting the squirts to 6 would have worked previously, as it sounds like you discovered. 3 was definitely broken though (As was 5).

noisymime, 2019-05-16:
Just incase you're wondering why 3 and 5 were broken, it comes down to how many degrees each requires between squirts and whether that can be done evenly within a single revolution. 

EG:
If you wanted 4 squirts per cycle, that's 1 squirt every `720 / 4 = 180` degrees, which is 2 squirts per revolution (`360/180`).
Same goes for say 6 squirts, which would be 1 squirt every `720 / 6 = 120` degrees, which is 3 per revolution (`360/120`).

BUT

If you wanted 3 per cycle, that's 1 every `720 / 3 = 240` degrees, which is 1.5 squirts per revolution. If you're only tracking the crank degrees up to 360, then things break when it tries to do 0.5 of a squirt.

V6Vintage, 2019-05-16:
Ok great, i will give it a go soon - 6 squirts alternating was indeed working BUT only up to a duty cycle of 15%, my "PwLimit = 90000" hard coded fudge has sorted that - completely the wrong way of doing it i know, but i'm an idiot and it works. the engine has never ran so smoothly! Thanks Josh

noisymime, 2019-05-16:
Cool. If you can let me know how you go, it would be good to confirm whether this is a closed issue or not

V6Vintage, 2019-05-18:
Hi Josh, I just warmed the engine up before loading the latest master but unfortunately the injector duty cycle stays at 0% throughout the cranking process, confirmed in MLV. Exhaust smelt dry. I tried 3 squirts simultaneous and two squirts simultaneous as i know that worked before.  I loaded my frankentune in and it fired back up as normal so am confident it's not a hardware issue.  I also tried shifting my req fuel higher & lower with no joy, thanks

noisymime, 2019-05-18:
Do you have a cam signal?

What pulse width / duty cycle were you seeing in TS?

V6Vintage, 2019-05-18:
I don't have a cam sensor, all i need is for both of my injectors to fire every 120 degrees i.e. 3 per rev, 6 squirts per cycle. This works but with a max 15% duty cycle. I can fit a cam sensor but if it doesn't sort the problem i'm going back to carburettors! haha

noisymime, 2019-05-18:
With the latest code (and ini file loaded in TS!!) what duty cycle are you seeing with 6 squirts per cycle?

V6Vintage, 2019-05-18:
0% everywhere - TS, MLV, real life, and yeah i load the ini with every update, 

hopefully a cranking log has attached and my current Tune. I can't attach a backed up tune file as it's 33mb.
cheers
[V6Vintage 6 Squirts no dutycycle.zip](https://github.com/noisymime/speeduino/files/3194046/V6Vintage.6.Squirts.no.dutycycle.zip)

noisymime, 2019-05-18:
Suspect there's something else in the new firmware causing you problem here. Unrelated to the 3 cylinder stuff. One thing I did notice is the secondary fuel table is enabled with all default values. Just try switching that off and see if it makes any difference. It's under Tuning->Second Fuel Table.

(All part of the 'fun' of using the master firmware)

V6Vintage, 2019-05-18:
ah yes it's running now! i hadn't even opened the secondary fuel table. so -this fuel table disabled gets the duty cycle 'working' again for me.

Just tested the max injector duty cycle whilst parked (ignition off.... stepper motor wired to the VR plug and spun with a cordless drill...!) and .... it all seems great!

Is this good enough for me to close this issue? Thank you for all your help

noisymime, 2019-05-19:
Excellent!

I think everything probably should be right to go now, so I will close this off. If there are any problems on the car though, just reply and I'll open it back up

closed 2019-05-19 by noisymime