speeduino/speeduino #171: add Sam MCU support

Autohome2, 2018-11-19:
Proposal to add Sam MCU support , samd/c21 offer scope for improvement over existing mega2560 in 64 or 100pin format.
The e53 versions offer compatible performance and features to the teensy 3.5 and are code and pin drop compatible to the c/d series.
Motor bike users will want to use the 32 pin packages for super small speedy ECU

turboedge, 2021-02-25:
Posted bounty. Specifically looking at SAMC21J for 4x4 channel support. Thanks!

noisymime, 2021-03-01:
That bounty is huge! Unfortunately for the SAMC21 is very likely going to be dependant on the addition of Arduino Framework support, which doesn't yet exist (See https://github.com/platformio/platform-atmelsam/issues/121).

SAMD21 is very possible, but obviously has a slightly different focus to the SAMC21.

turboedge, 2021-03-01:
I thought, naively, that the SAMD and SAMC were close enough to the same at the core that they would be more compatible with each other. This is unfortunate as the amount of hardware required to make the SAMD work with 5v signals and CAN is far greater than that of adding a USB interface, that could be off board even, to the SAMC.

Autohome2, 2021-03-01:
Is this not what you mean to add support to Arduino?
https://github.com/mattairtech/ArduinoCore-samd

The d51 and e53 are closer relatives that the c/d21 , with adafruit adding
Arduino support for d51 (like the grand Central board)

Darren

noisymime, 2021-03-02:
The `mattairtech` core is the closest thing I've seen, but hasn't been integrated or used by anyone. It's 2+ years out of date and doesn't look to be actively maintained any more :( 
See the comments on the PIO issue I linked to above.

There looks to have been some work done on it at https://github.com/prices/platform-atmelsam
But from what I can see it is far from complete (Though I haven't actually tried it yet).

Machine-Hum, 2021-03-06:
Any reason to not use the Atmel SAM3X8E? It's already integrated into the Arduino ecosystem?

noisymime, 2021-03-07:
The timer model on the SAM3X/A units is much more limited than what's on the SAMc/d/e MCUs.