speeduino/speeduino #179: Vbat voltage correction/offset

PSIG-dev, 2018-12-27:
Feature request of a voltage offset/correction, adjusted in TunerStudio, between a meter reading at the Speeduino input and the reported battery voltage by Speeduino.  Perhaps a simple slider to adjust until the voltage readings match.  No other curves or compensation.  This would allow corrections due to inherent error in Arduino internal voltage or regulator reference, but also allow offsets such as adding inline reverse-polarity diode/transistor protection with an offset for true system voltage, allowing proper application of voltage corrections (injector latency, coil dwell, etc).

Tobbera, 2019-04-27:
Good idea! I'm altering sensors.ino at the moment. 

`tempReading = fastMap1023toX(AnChannel[pinBat-A0], 162); //Get the current raw Battery value. Permissible values are from 0v to 24.5v (245)`

Replace 162 to match your true voltage.

noisymime, 2019-07-01:
Playing around with this at the moment and there's the question of whether the adjustment/offset should be % based or a fixed voltage offset. From the testing I've done, a fixed offset voltage seems to work well, but I thought I'd check and see whether there are any known use cases that would require a % adjustment instead.

Tobbera, 2019-07-05:
I have never used % adjustment. Static ADC read value is what I have had luck with. But there might be need for both in the case of a diode and voltage divider. First 0,5V off in the diode then percentage on the voltage divider,

PSIG-dev, 2019-07-05:
Agreed on the basis that most issues are based on general offset for the majority, and while drop may change with current, % would improve dynamic precision.  That would be gravy.  So either approach could work, with a basic offset value for measured difference under typical conditions; or a % that gives same result but (hopefully) remains correct with minor offset changes.  I think the first goal is a general correction to avoid most secondary issues and effects, but if % will help or not hurt (cause worse errors in certain circumstances) in achieving that goal, then go for it.

noisymime, 2019-07-08:
I've pushed a change which has this as an absolute correction value. Have a play and see if it works like that and if not comment and I'll open it back up to add a % variation as well

commit 3ee8e118 (2019-07-08): Add battery voltage offset calibration. Fixes #179
closed 2019-07-08 by noisymime