speeduino/speeduino #185: ALPHA - N CONIFG

ermancha, 2019-01-11:
Hi Guys, first a presentation my name is luis and i'am from Seville SPAIN i build a speeduino 0.4 board and runs over renault clio sport 172, my engine have a aftermarket CAMS and have serious problems with MAP at idle have very low depressure in the inlet.
I Want to speeduino uses only TPS not MAP (only for baro integrated map)
when i create a new project on tunerstudio i will put ALPHA-N
Then i import basemap
Then Change the config for Engine Constants & Spark for TPS and in this moment y have errors at open the project and the ball in the VE Table its freeze & Spark too.
The AFR Table reads the TPS

Now i create new project with speedDensity then config TPS in engine Constants & Spark
Now the VE Table ball runs fine with TPS & Spark Too
But the AFR Table its in MAP

I need to use only TPS what is the problem?
The Firmware is the OCT-18 same problem in AGO-18

all things i do.

new TS project,alpha N, TPS on engine constants & spark, save, restart, no warnings, all in offline,
open new arduino sktech, write blanc,
open speeduino oct ino upload
open last projet in TS (no Warnings)
conect speeduino
sincronize
the same problem
restart TS
warnings becomes

Tested on old firm DEC_16 and works perfect.

I think the problem its no similar data into ini of TS and the config inside de firmware

Please test the error.

Thanks, Best Regards.

ermancha, 2019-01-11:
Febraury -18 is the lastest firm that the Tables (ve, IGn,afr) runs with TPS.
May 18 & newer are wrong

PSIG-dev, 2019-01-12:
This issue is at https://speeduino.com/forum/viewtopic.php?f=11&t=2515&p=31187#p31108.  Obvious effect is in Alpha-N fueling algorithm, VE table and Spark table do not appear to respond to TPS input.

noisymime, 2019-01-12:
The newer firmwares change the Y axis of these tables to be a 'Fuel Load' value rather than simply a sensor value. The logic is currently that if the RPM = 0, then the fuel is 0 and hence there is no fuel load. This is why you're not seeing the dot move on those tables when you don't have an RPM signal. 

I understand that this is not obvious though, so I'll look at changing this so that these change when RPM = 0.

PSIG-dev, 2019-01-12:
@LAV1000 has posted images in the linked thread showing no response on 2D display of VE and Spark tables, when 3D display shows properly **with** RPM?
https://speeduino.com/forum/download/file.php?id=4313

commit b0d02b3f (2019-01-15): Move the fuel load calculation outside of the sync/RPM check. Closes #185
closed 2019-01-15 by noisymime