speeduino/speeduino #191: Switching from Alt to Sim halves the pulse width

Old-Grey6, 2019-01-28:
Reqd Fuel - 10
PW 10ms
Squirts - 2
Alternating
4-stroke
4 cyl
4 INJs
Paired
I switch to Simultaneous, and PW halves(5ms).
They both have the same # of INJ pulses, just Alt offsets ch#2 by 180º, yet Sim halves the PW.
Test is here - https://youtu.be/sYio0XROS1I
And it produces this -
https://www.mediafire.com/convkey/3a0b/qnebtx4usft2hqu6g.jpg
Like PSIG says, this is a bit of a dangerous situation. Like one poster on the forum switched from Alt to Sim, but luckily didn't go for a drive after it started idling rough. He didn't know why, and I don't think he had O2, so it was potentially bad.
Thread - http://speeduino.com/forum/viewtopic.php?f=21&t=2556

Is there an undocumented convention, or is this a bug?

brunob45, 2019-01-28:
This is a convention.
2 squirt alternative is like : 
_*____/_*____/
____*_/____*_/
and 2 squirt sim is like :
_*__*_/_*__*_/
_*__*_/_*__*_/
where (*) is an injection event and / is the end of an engine cycle
As you can see, simultaneous make the injections happen twice as much, so we cut the PW to compensate

Old-Grey6, 2019-01-28:
Well then that's a FW bug, because Alt will continue to squirt 2 pulses when set on 2 squirts, or from the standpoint of Sim, Sim does not double the # of squirts.  Also there is not a documented warning in the Wiki telling you you MUST change to 1 squirt when using Alt, or double squirts in Sim. And even further it won't let you select 1 squirt in Alt anyway.

Old-Grey6, 2019-01-28:
Well then that's a FW bug, because Alt will continue to squirt 2 pulses when set on 2 squirts, or from the standpoint of Sim, Sim does not double the # of squirts.  Also there is not a documented warning in the Wiki telling you you MUST change to 1 squirt when using Alt, or DOUBLE squirts for Sim. And even further it won't let you select 1 squirt in Alt anyway.

brunob45, 2019-01-28:
I was meant to say :
2 squirt alternative is like :
..^...../..^...../
.....^../.....^../
and simultaneous :
..^...^./..^...^./
..^...^./..^...^./
where (^) is an injection event and / is the end of an engine cycle.
And yes, it looks like a FW bug.

Old-Grey6, 2019-01-29:
Yeah I think you're right.
I think what's supposed to happen is when on 2 squirts and you switch to ALT, it's supposed to automatically switch to 1 squirt. That would fall in line with MS's manual, and would indicate Speedy has a FW bug.

runesmdk, 2019-02-04:
See it as the "number of squirts" is the times you want either one(or all) injector channels to open on a 720degree crank rotation.

Old-Grey6, 2019-02-04:
Yeah I know, but if you switch to Seq Alt 2 squirts it internally changes it to 1 squirt. With Seq, Alt 2 squirts and Sim 1 squirt do the same thing. ie output 1 squirt, it's just Paired that does the half width weirdness.

PSIG-dev, 2019-03-07:
Bump.  All options should work properly, and should be consistent as we add yet more cylinders.  Best fuel distribution settings are necessary for both running and tuning more cylinder counts and features.  This is a basic requirement today, so I'm bumping it. ;-)

brunob45, 2019-03-08:
Helo guys! I just opened a PR ([#211](https://github.com/noisymime/speeduino/pull/211)) with the beginning of a solution.
Hope it helps!

PSIG-dev, 2019-04-11:
From the thread linked earlier: **_"TL;DR — the calculated and actual fuel flow should remain the same when changing either the mode or the number of injections."_**  This is important, and why I am bumping it.  Also bumping #211.  Thanks for the efforts to resolve this!

brunob45, 2019-05-28:
Just pushed some changes (see PR #211).
I was aiming too high, and began a too big rewrite of the existing code.
Now, I went back and fixed only what needed fixing.
The update function still needs testing, but the init part is good to go.

brunob45, 2019-07-06:
To validate that the fix I am proposing works, I created some scripts to let us run and test the speeduino code on a computer (see my branch [testing](https://github.com/brunob45/speeduino/tree/dev/testing)).
I started by setting the basic constants (reqFuel, nCylinders, nSquirts, alternate, injLayout) in the simulated EEPROM, then I ran initialiseAll(). The results were :
```
reqFuel:12, nCylinders:4, nSquirts:2, alternate:FALSE, injLayout:PAIRED
req_fuel_uS:3000
CRANK_ANGLE_MAX_INJ:360
channel1Inj:0
channel2Inj:0
channel3Inj:disabled
channel4Inj:disabled

reqFuel:12, nCylinders:4, nSquirts:2, alternate:TRUE, injLayout:PAIRED
req_fuel_uS:6000
CRANK_ANGLE_MAX_INJ:360
channel1Inj:0
channel2Inj:180
channel3Inj:disabled
channel4Inj:disabled
```
While the only setting that changed was "alternate", the total amount of fuel (req_fuel_uS) changed.
I will apply the same tests on the fix I proposed in May a bit later.

brunob45, 2019-07-07:
With my fix, the values will be : 
```
reqFuel:12, nCylinders:4, nSquirts:2, alternate:FALSE, injLayout:PAIRED
req_fuel_uS:6000
CRANK_ANGLE_MAX_INJ:360
channel1Inj:0
channel2Inj:0
channel3Inj:disabled
channel4Inj:disabled

reqFuel:12, nCylinders:4, nSquirts:2, alternate:TRUE, injLayout:PAIRED
req_fuel_uS:12000
CRANK_ANGLE_MAX_INJ:720
channel1Inj:0
channel2Inj:360
channel3Inj:disabled
channel4Inj:disabled
```
As one can see, the amount of fuel injected over a full engine cycle (`req_fuel_uS/CRANK_ANGLE_MAX_INJ`) stays constant.

When updating, to get the same results as above, we would need to divide reqFuel by 2 when NOT alternating:
```
reqFuel:6, nCylinders:4, nSquirts:2, alternate:FALSE, injLayout:PAIRED
req_fuel_uS:3000
CRANK_ANGLE_MAX_INJ:360
channel1Inj:0
channel2Inj:0
channel3Inj:disabled
channel4Inj:disabled
```
or multiply nSquirts by 2 when alternating :
```
reqFuel:12, nCylinders:4, nSquirts:4, alternate:TRUE, injLayout:PAIRED
req_fuel_uS:6000
CRANK_ANGLE_MAX_INJ:360
channel1Inj:0
channel2Inj:180
channel3Inj:disabled
channel4Inj:disabled
```

brunob45, 2019-07-07:
When speaking abour fully sequential injection, the constants will no longer have any effect on the actual injection :
```
reqFuel:12, nCylinders:4, nSquirts:2, alternate:FALSE, injLayout:SEQUENTIAL
req_fuel_uS:12000
CRANK_ANGLE_MAX_INJ:720
channel1Inj:0
channel2Inj:180
channel3Inj:360
channel4Inj:540

reqFuel:12, nCylinders:4, nSquirts:2, alternate:TRUE, injLayout:SEQUENTIAL
req_fuel_uS:12000
CRANK_ANGLE_MAX_INJ:720
channel1Inj:0
channel2Inj:180
channel3Inj:360
channel4Inj:540

reqFuel:12, nCylinders:4, nSquirts:4, alternate:TRUE, injLayout:SEQUENTIAL
req_fuel_uS:12000
CRANK_ANGLE_MAX_INJ:720
channel1Inj:0
channel2Inj:180
channel3Inj:360
channel4Inj:540
```