speeduino/speeduino #193: Request:  Percent dwell setting

PSIG-dev, 2019-02-02:
Most manufacturers used current-control or dwell-control modules at one time or another, GM HEI, several Bosch modules, Ford TFI, etc.  In order to control dwell properly, they need to function as a percentage of available dwell time.  

The request is for a Speeduino function and TunerStudio setting to allow dwell percent as-opposed to fixed dwell.  User-specified percent would be great, but either a fixed Maximum Available setting (preferred), or fixed 50% as used in another system would appear get the job done in most cases.

ric355, 2019-02-04:
Can you define what "available dwell time" is here - is it based on current RPM?

PSIG-dev, 2019-02-05:
Correct.  For example using 50% dwell; if current available time between coil firings is 4ms, dwell would begin at 2ms from end of last dwell.  Note this is not _after_ spark discharge, but including it, effectively shortening discharge in extreme cases.  Also note that a Fixed 50% is least preferable, as it would (in many cases) diminish spark energy to a greater degree as RPM rise, making User-Defined % or Maximum Available % preferable.  The goal is to allow OEM current-limiting or auto-dwell modules to function as-designed, and to provide maximum spark energy.