speeduino/speeduino #195: Issue in Injection signal

WirelessHubMaster, 2019-02-05:
I have tried to map the pins  to the MEGA 2560 as per the case 3 (0.4 shield), kept the code the same as the original. pinInjector1 signal is being displayed, the other signals are at 0. I've tried everything, from resetting the Board to uploading and changing the wires. It still isn't working. Note: I'm Using Analog Discovery as my oscilloscope.

[image]

noisymime, 2019-02-05:
Post your tune please. Very likely this is something misconfigured within there.

WirelessHubMaster, 2019-02-06:
Is this the tune you asked for? Please let me know if you want anything else
[Injectortrial1.zip](https://github.com/noisymime/speeduino/files/2835118/Injectortrial1.zip)

noisymime, 2019-02-06:
Your tune is set for 1 cylinder / 1 injector, so none of the other injector outputs will be active. This is the normal, expected behaviour.

WirelessHubMaster, 2019-02-07:
I'm going for 1 Cylinder/ 4 Injectors as the main project. I've added the injection timing cases in an loop(subcases) , the signals should be present as 1, 1a, 1b, 1c. Also how and where should i make changes in speeduino.ini to change the phase of the injector 1 and the remaining three pulses so that the three signals ( 1a, 1b,1c ) that i'm trying to implement have a delay/Phase shift from the main Injector1 pulse ?

noisymime, 2019-02-07:
Single cylinder with 4 injectors is certainly a unique setup!
So you want all 4 injector outputs to fire at defined intervals.

If so, you're probably best to configure it as a 4 cylinder engine, then change the timing between the pulses by setting it as odd-fire and defining the angles you want.

WirelessHubMaster, 2019-02-07:
Sorry for the confusion. It's a single cylinder with single injector with multiple timing (4 pulses) for fuel injection. I'm currently unable to configure it from the .ini file, hope you can guide me through for the above.

noisymime, 2019-02-08:
Have you tried increasing the number of squirts setting?

WirelessHubMaster, 2019-02-08:
Doesn't allow multiple squirts for single cylinder/Single Injector. Tunerstudio gives a warning.

Magnum828, 2019-02-17:
This is also a problem for me, I think the calculation should be that you can't alternate e.g. 2 cyl with 3 squirts because that isn't providing a whole number. But 1 cyl 1 inj and 2 squirts per cycle should be possible!

brunob45, 2019-10-17:
The {number of squirts} value sent to speeduino is not the number we see, but instead {number of cylinders} / {number of squirt}. 1 cylinder and 4 squirts would mean a value of 1/4=0.25: That number must be an integer, so this configuration is invalid. This is a limitation hardcoded into TunerStudio, as Megasquirt was coded this way. Unfortunately, your setup won't be possible as long as we're using the default Engine Constants panel.

noisymime, 2022-10-03:
This limitation was removed and the issue resolved with 8886d3dbd75dbbcd983795a6705d9fae0a813242

closed 2022-10-03 by noisymime