speeduino/speeduino #198: Hard limiter method: Rolling

ric355, 2019-02-12:
I was looking at how the rev limiter works, and I can't see how it will ever cut the ignition when configured for Rolling Cut. This is because the line of code which calculates which ignition channel to cut is commented out.

`        if(configPage2.hardCutType == HARD_CUT_FULL) { ignitionOn = false; }
        else 
        { 
          if(rollingCutCounter >= 2) //Vary this number to change the intensity of the roll. The higher the number, the closer is it to full cut
          { 
            //Rolls through each of the active ignition channels based on how many revolutions have taken place
            //curRollingCut = ( (currentStatus.startRevolutions / 2) % maxIgnOutputs) + 1;
            rollingCutCounter = 0;
            ignitionOn = true;
            curRollingCut = 0;
          }
          else
          {
            if(rollingCutLastRev == 0) { rollingCutLastRev = currentStatus.startRevolutions; } //
            if (rollingCutLastRev != currentStatus.startRevolutions)
            {
              rollingCutLastRev = currentStatus.startRevolutions;
              rollingCutCounter++;
            }
            ignitionOn = false; //Finally the ignition is fully cut completely
          }
        } 
`

It looks like "curRollingCut" will always be zero, and therefore none of the ignition channels will be cut;
For example, channel 2:
`            if( (ignition2StartTime > 0) && (curRollingCut != 2) )
            {
              setIgnitionSchedule2(ign2StartFunction,
                        ignition2StartTime,
                        currentStatus.dwell + fixedCrankingOverride,
                        ign2EndFunction
                        );
            }
`

and so on.

Has it been deliberately disabled or have I misunderstood how it works?

noisymime, 2019-02-12:
The method for this was changed back in July and the previous rolling method had proved to be insufficient for actually stopping revs from climbing. The term 'rolling' is misleading with what this changed to as the cut no longer rolls between ignition channels (as you've pointed out). What it does is do a full cut of the ignition system (`ignitionOn = false;`) but only for a fixed number of revolutions rather than until the RPM has dropped below the hard cut limit. 

This seems to do a better job at keeping the RPM closer to the limit. 

Honestly, the rev limiter algorithms could use some improvement. Currently they do work, but tend to be too hard at cutting revs, causing them to drop too much. The rolling cut method varied in success based on factors whether you were running sequential or wasted spark ignition etc. I'm sure there's a better way, but I haven't looked into it further so far.

ric355, 2019-02-12:
Thanks for the quick reply. I see I didn't spot the ignitionOn in the rolling cut section.

I haven't quite hit the rev limiter on my engine yet so don't know what it feels like. I will lower the limit and see how it behaves, and maybe have a play with a few algorithms.  I was actually looking to see how I might integrate a traction control signal in, so I could encourage the ECU to cut based on an external input.

ric355, 2019-02-12:
Some interesting reading here;
https://www.w8ji.com/rpm_limiters.htm

closed 2019-03-13 by ric355