speeduino/speeduino #200: Inverted PWM option for Boost control

ExperionBw, 2019-02-17:
This is a feature request for the option to invert the PWM output for boost control.
What I mean by this is: instead of Duty cyle being high to get on boost, then lowering to limit boost,
It needs to be 0%Duty cycle when not in boost and it then needs to rise when boost comes in to limit boost.

The reason I need this, is because Im planning on using a VGT turbo which is controlled with vacuum instead of boost.
when the vane actuator of such a turbo is in complete vacuum (0% Duty Cycle) it creates the largest amount of restriction (increasing exhaust gas speed)which will spool the turbo fastest.
Then when Duty cycle increases the vanes will decrease restriction, lowering exhaust gas speed and limiting boost.

If I were to do some 'presure inverting' by 'leaking' vacuum in with a solenoid it would allow me to use boost control in the normal way, but it would mean that the idle position of the vane actuator is in the least restricted position and when I apply throttle the vane actuator would need to move which would take some time and also negatively impact throttle response.

In conclusion: I hope its not too difficult to get this feature added, I will put a small bounty on it :)

ExperionBw, 2019-02-17:
Bountysource's search system seems to be broken so I cant find this issue on there, Il try again in a few hours

ExperionBw, 2019-02-23:
I want to add that there is a 25$ bounty on this now

UnicodingUnicorn, 2019-08-12:
I'm working on a solution for this now, but may I suggest as a stop gap I think you probably can put the output through a NOT gate.