speeduino/speeduino #202: Settings Changed that require a power cycle to take effect

WirelessHubMaster, 2019-02-22:
Hey Guys, I'm trying to remove the reset set on the Arduino board such that every input given at any point of time will have a smooth transition to the output. 
Note
1) I've tried Shorting the pin on the board
2) asm volatile ("jmp 0"); - I've used this in the speeduino-master.ino file , the code compiles but no effect is observed whatsoever.
3)Commenting the 'reset control' and 'reset control pin' in the speeduino.ini file.

Please suggest an alternative method to do the same

ric355, 2019-02-22:
The message you have quoted is nothing to do with reset control. There are a bunch of definitions in the speeduino ini file which determine which settings need you to restart the controller for if you change them. When any one of those is changed you will get the message you quoted.

A lot of the config is processed during initialization so that it does not have to be recalculated on the fly. This speeds up processing.  If you wanted to get rid of that message so that any change took effect immediately, you would have to make many many changes to the firmware, and you would introduce an ongoing maintenance effort to keep up with all of the changes being made.

WirelessHubMaster, 2019-02-22:
Thanks for the response! Is it not possible just for testing for injector squirt time check? I want to inject my fuel in micro seconds thus trying to alter the reset so that I can give continuous input to the Arduino and the Arduino won't just shut down.

ric355, 2019-02-22:
Injector open time can be changed in real-time, but the squirts per cycle cannot and there are a lot of firmware changes needed to support the latter.

Basically, anything that produces that message does not have any code in the firmware to support real-time changes.

noisymime, 2019-03-15:
Closing this as generally all settings that need to be live editable (and can be without excessive checks in the code) are this way already.

closed 2019-03-15 by noisymime