speeduino/speeduino #203: 201902 Dwell voltage compensation possible 255 roll over issue

ESDave05, 2019-02-23:
SW Version: 2019 Feb Release
Test Environment: Bench - "JimStim" simulating crank / cam reference, pots for sensor inputs etc...

Preconditions:
[image]

[image]

As i ramp voltage up and down in crank I see some bizarre behavior with the dwell:

[image]

noisymime, 2019-02-24:
Thanks for the report. 

I'll check this out ASAP

noisymime, 2019-03-13:
I've confirmed that there is a bug in how the dwell is reported in TunerStudio (And the log) if it exceeds 25.5ms. The unit itself will still continue to output dwell times over 25.5, but the number in TunerStudio is looping. 

This is very much an edge case as dwell times over 25.5ms are fairly unrealistic, but I will make sure that this is reporting correctly going forward. Again, i'd stress that the actual ignition output will still dwell at the correct time, it's only the value displayed in TS that is wrong

commit 4f4781a2 (2019-03-15): Send dwell value as 16 rather than 8 bit to TS. Fixes #203
closed 2019-03-15 by noisymime