speeduino/speeduino #205: No fuel pump priming pulse if you have USB connected prior to starting car

ESDave05, 2019-02-24:
This is probably getting a bit pedantic but just an observation:

My typical workflow for tuning the MX5 would be:

Key out
Connect laptop to Speeduino, start tuner studio
Turn ignition on
Check sensors for valid readings etc...
Start Log
Start Engine

I notice i don't get the 5 second long fuel prime - i havn't dug into the code but I would assume it's in some initialization code that run when the USB was first connected - from there the Arduino does not go through a reset with key on / off.

Easy enough work around, just an observation. I often don't work on my car for months at a time so I would imagine fuel lines need a decent prime - a poor start due to no prime could be mis interpreted as a red herring tuning issue etc...

ric355, 2019-02-24:
Indeed this is because the arduino gets powered up when the USB line is connected, and hence the initialisation happens before you give power to the fuel pump via ignition. 

As speeduino does not have an ignition on input, it isn't possible to move the priming pulse out of the initialisation code, although I suspect it would be a relatively easy mod and firmware change to achieve it.

noisymime, 2019-02-24:
Yeah, this is working 'as designed' as the intent is to prime only when the unit is powered on.

That said, this comes up quite often as a question, so I think it might be time to change that behaviour. I'm going to mark this as a feature request and look at changing it shortly, it should be very straightforward.

brunob45, 2019-02-25:
[brainstorming] A potential solution would be to start the fuel priming only when the speeduino first sees at least 12v on the battery voltage.

noisymime, 2019-02-25:
Yeah, the check itself for this is relatively straightforward, in basic pseudo code:

`if $current_voltage < 5.5 and $new_voltage > 7 then prime_pump()`

This would be in addition to the existing startup priming routine. Basically if the voltage was less than 5.5, the system was likely to have been on USB. If the voltage has now gone above 7 (The minimum dropout voltage for the regulator) then it's likely that the ignition has been turned on. 

There's a little more error checking logic to add, but this is the high level of it.

ric355, 2019-02-25:
There are some other events that are also associated with key-on such as stepper homing. Would the intent be to include those as well?  Stepper is a good example of where the system does not behave properly if you connect the USB first, as it'll carry out the homing event without moving the stepper given no 12v is present. This in turn gets the stepper position immediately out of sync.

My concern about using input voltage is that transient voltage drops could generate false positives and cause the priming event (and any others we associate with it) to happen. Since the end of the priming event is to switch off the fuel pump we'd have to be careful to ensure this doesn't happen if a transient drop occurs while the engine is running.

Perhaps put it on a TS setting so that anyone who has problems can disable it.

noisymime, 2019-02-25:
Adding an RPM check definitely makes sense if the stepper is being added, which seems like a reasonable idea.

Even with the voltage though, if you're checking that it was previously <7v, then that generally should be sufficient. If the system voltage was below 7v and there wasn't USB connected, then the unit has very likely reset anyway due to the regulator dropout voltage being about 7v. 

With the RPM check though (ie only do this if `rpm == 0`) then it should be safe.

ric355, 2019-02-28:
I have a prototype available for this one; I've done a bunch of bench testing on it and am going to test on the car later today. I should be able to submit a pull request in few days if you wish. No problem if you've already worked on it.

noisymime, 2019-02-28:
It's not fully tested, but I just pushed up the basic stuff that I put together around this. It seems to work, but it can be dependant on the filter value of the battery ADC, so I need to keep having a bit of a play around

ric355, 2019-02-28:
My solution is here:

https://github.com/ric355/speeduino/commit/5984e7fd98da23f2045d75edcab6e1ce185f7750

I also added an output channel so you can see the stepper priming event in TS using an indicator;

https://github.com/ric355/speeduino/commit/dd031771a028021964309f4263d797e7c245e298

I successfully tested it on the car today.

I did a similar thing regarding the battery threshold, but I did it in a different place and did not use a gap (your 55 to 70). I just said if it goes from < 70 to >= 70 then that is the trigger. This is because the ADC filtering setting has a big effect on the values you see within readBat as the voltage rises.

ESDave05, 2019-03-04:
Just an interesting observation, i actually get a phantom prime during ignition transition for 350ms.

I think it might be an RPM glitch detected as engine running, but RPM is flatline in datalog. Happens in the car and on the bench.

[image]

ToxicAvenger, 2019-05-03:
@ESDave05 
Did you solve your problem with phantom prime pulses?
I'm observing something similar in my setup. After configuret period the pump goes off. Then it turns on again and turns off. I hear, that relay is clicking. The rpm log in TS is flat.

ESDave05, 2019-05-05:
I'm happy to keep it as a "feature" :) but for me it only happens once on Ignition transition.

turboedge, 2022-05-04:
Was this ever added to the firmware? I've had reports of it also playing heck with stepper idle valve homing.

noisymime, 2022-05-05:
Yeah, this has been in the firmware since e8c31d5ce743b3502395b9c48d6324dd6ecf5c78 though only for the fuel pump priming, not intending to cover stepper homing etc. 

Worth nothing is that it is somewhat hardware dependent as well, requiring the USB line to backfeed onto the main power line when 12v is not connected. Most basic LDOs will do this, but not all and other power regulator designs (Eg switching regulators) will not necessarily work.

noisymime, 2024-03-15:
Closing as no further action is intended on this.

closed 2024-03-15 by noisymime