speeduino/speeduino #209: Request: Trigger decoder For Fiat Punto HGT 1.8 - Fiat Barchetta - Fiat Coupe 1.8 16V

AntonisPagratis, 2019-03-05:


lsdlsd88, 2020-04-17:
Hello I have tried writing some code for this trigger but is anyone available to test it on a real car or bench? with ardu-stim seem to work. I cannot get hold of Antonis through the forum PMs.

kovama1, 2021-01-22:
> Hello I have tried writing some code for this trigger but is anyone available to test it on a real car or bench? with ardu-stim seem to work. I cannot get hold of Antonis through the forum PMs.

Hi,
I will be doing speeduino conversion for Punto MK1 soon, so I will be probably able to test it ;)

lsdlsd88, 2021-01-22:
cant seem to find any option to message you here so join slack or the forum plz :)

Edjaime, 2021-08-16:
> Hello I have tried writing some code for this trigger but is anyone available to test it on a real car or bench? with ardu-stim seem to work. I cannot get hold of Antonis through the forum PMs.

I can test. You can send to me? I need urgent. Please

Edjaime, 2021-08-16:
Any news? I really need it too!

lsdlsd88, 2021-08-16:
message me on speeduino forum.

Edjaime, 2021-08-17:
> message me on speeduino forum.

I am unable to post messages via the speeduino forum. I think it's because I'm new to the forum. My username is Edijaime. Help me please!

Edjaime, 2021-08-17:
> message me on speeduino forum.

I sent a message to your email