speeduino/speeduino #213: DFCO after start delay

ric355, 2019-03-13:
Last week I was doing some work on my tune, and seemed to be having trouble getting the car to idle after start - it just stalled. When I looked at the datalog I realized that decel fuel cut was kicking in immediately after start because the revs were rising above the cut off point I had set (1400rpm) and the throttle was at zero percent. This of course caused the fuel to cut and the engine to stall as the engine was cold.

I was thinking that maybe we could add a setting to disable fuel cut for n seconds after engine start. This would give chance for revs to settle after start. It's mainly an issue that will occur during tuning and of course DFCO can be switched off, but it might help reduce frustration in some circumstances. I was thinking the default should be something like 15 seconds.

If you think this is a useful addition let me know and I'll write the code for it.

runesmdk, 2019-03-22:
Usually, setting a coolant temp limit on dfco also fixes this issue

VitorBoss, 2020-05-29:
@noisymime This is fixed by recent functions added, you can close it.

noisymime, 2020-05-29:
Feature added in 558d063f56572807eff691b382cc11c05ca4361b

closed 2020-05-29 by noisymime