speeduino/speeduino #217: New CRC32 implementation breaks teensy build

brunob45, 2019-04-04:
The new implementation of CRC32 was not ported to FastCRChw.ino, so the project won't compile for the teensy controller since commit 49a6744e36c74d2c9ba29159e9b862a16b80a748

noisymime, 2019-04-04:
Not sure how I missed that. Yeah the Teensy is being pushed to use the HW CRC rather than the SW implementation, which doesn't have the changes I added in. 

I'll get this resolved ASAP

TradieUte, 2019-04-06:
Have found a solution.

In utils.h, added code -

#if defined(KINETISK)
#define CRC32_M  CRC32.crc32(&raw_value, 1)
#define CRC32UPD_M  CRC32.crc32_upd(&raw_value, 1)
#else
#define CRC32_M CRC32.crc32(&raw_value, 1, false)
#define CRC32UPD_M CRC32_val = CRC32.crc32_upd(&raw_value, 1, false)
#endif

In utils.ino -

changed every instance of 
CRC32.crc32(&raw_value, 1, false)
to
CRC32_M

changed every instance of 
CRC32_val = CRC32.crc32_upd(&raw_value, 1, false)
to
CRC32UPD_M 

In FastCRC.h line 139, changed code as follows

#if defined (KINETISK)
  uint32_t crc32(const uint8_t *data, const uint16_t datalen);
#else
  uint32_t crc32(const uint8_t *data, const uint16_t datalen, bool reflect=true);		// Alias CRC-32/ADCCP, PKZIP, Ethernet, 802.3
#endif
  uint32_t cksum(const uint8_t *data, const uint16_t datalen);		// Alias CRC-32/POSIX

#if defined(KINETISK)
  uint32_t crc32_upd(const uint8_t *data, uint16_t len);
#else
  uint32_t crc32_upd(const uint8_t *data, uint16_t len, bool reflect=true);			// Call for subsequent calculations with previous seed
#endif

With case correction - issue #218, code now compiles for teensy

brunob45, 2019-04-06:
The code compiles, but will the serial command 'd' still work? I think the argument 'reflect' should be used with all controllers

TradieUte, 2019-04-06:
Try it out and post the result. The changes made direct the code to use hardware CRC when compiled for teensy. It was assumed that fastCRChw.cpp (teensy) and fastCRCsw.cpp (non teensy) contained no errors.

brunob45, 2019-04-07:
[image]
I got this error after updating my code with your fix.

brunob45, 2019-04-07:
Began working on this issue on my branch [dev/teensyfix](https://github.com/brunob45/speeduino/tree/dev/teensyfix).
It also fixes #218 and another warning in utils.ino line 171.

brunob45, 2019-04-08:
I made a pull request earlier today (#219), in which the code now behave as intended. No more warnings or CRC errors from tunerstudio. What result are you talking about?

TradieUte, 2019-04-08:
Received the same error when initially loading starting T/S with Teensy, but could not create when reloading tune. Tried again using mega2560 and got similar result. Cleared Mega EEPROM, recompiled unmodified code and connected T/S, with new T/S project. Got same error. 
T/S log attached. (Logs for modified code similar, wrt CRC errors.)
[TunerStudioAppDebugMega.txt](https://github.com/noisymime/speeduino/files/3052202/TunerStudioAppDebugMega.txt)

Seems like an issue with CRC algorithm.

noisymime, 2019-04-08:
@TradieUte Those errors aren't CRC related, see just about them you can see a whole bunch of messages like `CRC matches for page 10:x59 xAD x15 x7D` which shows that the CRC is working fine. 

Not immediately sure what those CAN ID errors are related to though, I've not seen anything like them before. I'll see if I can reproduce

TradieUte, 2019-04-08:
Issue with Mega occurs after clearing EEPROM.
Downloaded latest Speedy code (Speeduino_master) from github, compiled and loaded Mega, no errors connecting with T/S.
Checked all new Speedy files against released. Speeduino_master vs Speeduino_201903. No difference apparent.
Scratched head.
Cleared Mega EEPROM, loaded latest compiled code, connected to T/S, got the error.
[image]
[TunerStudioAppDebug.txt](https://github.com/noisymime/speeduino/files/3052334/TunerStudioAppDebug.txt)

closed 2019-04-08 by noisymime

TradieUte, 2019-04-08:
Issue with Mega occurs after clearing EEPROM.
Downloaded latest Speedy code (Speeduino_master) from github, compiled and loaded Mega, no errors connecting with T/S.
Checked all new Speedy files against released. Speeduino_master vs Speeduino_201903. No difference apparent.
Scratched head.
Cleared Mega EEPROM, loaded latest compiled code, connected to T/S, got the error.

TradieUte, 2019-04-08:
Have Teensy CRC running finally. Analysis of CRC errors showed that the values returned from the controller were inverted (reflected?). Code in utils.ino used manual inversion on all page data that were in error.
Corrected all instances using:-

      //Do a manual reflection of the CRC32 value
//change
#if !defined(KINETISK)
      CRC32_val = ~CRC32_val;
#endif
//end change

No more errors.

PR #219 (2019-04-08): Fix teensy compilation errors and CRC32 calculation
This PR fixes #217 and #218.