speeduino/speeduino #221: TS Command "P9L" does not terminate

TradieUte, 2019-04-16:
line 1263 of comms.ino contains an invalid compare.

Original (commented) line and change follows:-

//          for (pnt_configPage = &configPage9; pnt_configPage < ((byte *)pnt_configPage + 128); pnt_configPage = (byte *)pnt_configPage + 1)

          for (pnt_configPage = &configPage9; pnt_configPage < ((byte *)&configPage9 + (EEPROM_CONFIG9_END - EEPROM_CONFIG9_START)); pnt_configPage = (byte *)pnt_configPage + 1)

noisymime, 2019-04-16:
Thanks for the report, I'll fix shortly. 

I'll likely use the `npage_size` array rather than the EEPROM storage locations, but it will ultimately. give the same result.

noisymime, 2019-04-16:
Assuming you're referring to this line: https://github.com/noisymime/speeduino/blob/master/speeduino/comms.ino#L1218

It's worth pointing out that this should only be affecting the `useChar` version, which is the one that prints out the table values in human readable format. It's not the call used by TS.

TradieUte, 2019-04-16:
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/noisymime/speeduino/issues/221#issuecomment-483511884",
"url": "https://github.com/noisymime/speeduino/issues/221#issuecomment-483511884",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2019-04-16 by noisymime

TradieUte, 2019-04-16:
Have reviewed the change and fear I may have confused the real issue by emphasizing the change in age size.
The issue is that the pointer comparison pnt_configPage < ((byte *)pnt_configPage + 128) never returns false as pnt_configPage is continually incremented.
The clause should read pnt_configPage < ((byte *)&configPage9 + 128)

noisymime, 2019-04-17:
I really shouldn't code without coffee :D

You're 100% right, I'll fix shortly (Though it definitely should be 192 rather than 128, though that is taken care of with the nPage_size[] reference)

commit f01783a2 (2019-04-16): Use correct page size for useChar with canbusPage. Fixes #221