speeduino/speeduino #225: Request: Build in 1 or 2 spare analog inputs

ESDave05, 2019-04-30:
Hello all,

I am in a situation where I would like to add *1 fuel rail pressure sensor to my speeduino.

It seems like there are two ways to achieve this:

1. Modify the code to use one of the many spare A2D's... then I would have to re-incorporate these changes any time the base firmware is released and i want to upgrade.

2. Add a CAN or Serial expander device to the speeduino.

I am leaning towards 2. as the "proper" solution, but this is a lot of effort to just add *one Analog input to an ECU that has many channels free.

Request is: Could we bake in 1 or 2 spare ADC channels into the base Speeduino firmware? If the just the raw A2D decimal could be output I believe scaling, offset and units can be handled in the Tunerstudio config?

For the question of appropriate filtering - for the case where this sensor is not directly used for any part of the engine management algorithms I would be happy if it's just abit "noisy" in the datalog - or I would add a HW filter. I guess the fancy way would be some sort of moving window filter with a co-efficient you could change from TS??

noisymime, 2019-04-30:
It's still somewhat rough around the edges, but there is the (local) Aux input functionality that should largely cover this. You select an input as being analog, give it a name and assign it a pin. The values you get from it will just be the raw 0-255, but as you say, you can then scale them however you like on a gauge in TunerStudio

noisymime, 2019-04-30:
(This is under Accessories-Local Auxiliary Input Channels Configuration in Tuner Studio)

Autohome2, 2019-04-30:
Look at the information about serial3( secondary serial) in the wiki for serial expansion.
Also look in third party boards at my gpio as that offers  io expansion and can via serial3
Aux inputs can be configured to use both local analog and digital pins and remote pins.