speeduino/speeduino #227: No 360 degreee sequential injection on 3-cylinder 4.stroke

Tobbera, 2019-05-06:


Tobbera, 2019-05-06:
When trying to set up TS to run 3 cylinder 4 stroke semi-sequential, all injectors are firing at the same time. 

Please see this forum thread for examples and case explanation:
[](https://speeduino.com/forum/viewtopic.php?f=19&t=2717&start=30)

I think the problems lies within init.ino beginning line 316 and more specifically line 341. 

When choosing 2 stroke everything looks fine.

Tobbera, 2019-05-06:
Another related issue is that TS "stops" you from making certain choices. Like number of squirts etc. It pops up a message saying you need to have compatible numbers. This can be worked around by ignoring it and burning anyway.

Boogmeister, 2019-05-06:
> Another related issue is that TS "stops" you from making certain choices. Like number of squirts etc. It pops up a message saying you need to have compatible numbers. This can be worked around by ignoring it and burning anyway.

This is due to TS not allowing it.

This is due to restrictions in MS.

Nothing Speedy side of things to do as it's a limitation of TunerStudio itself.

Tobbera, 2019-05-06:
> This is due to TS not allowing it.
>
> This is due to restrictions in MS.
>
> Nothing Speedy side of things to do as it's a limitation of TunerStudio itself.

Yes, I have sent an email to EFIanalytics about it. But the main issue still persists.

LPG2CV, 2019-05-06:
I think 3 (also 5) cylinders is the issue here. 
You can't have Sequential as you don't have a cam signal. Semi-sequential and Paired need to halve the number of cylinders, so that can't work. But you don't have another option!

Tobbera, 2019-05-06:
I think we are a bit confused about the terminology. What I want to achieve is sequential within the 360 degree cycle, as I don't have a cam sensor.

Going two stroke in TS seems to be the only way of making this happen. All other settings are making very weird results. Have a look at the logic scopes in the thread.

Boogmeister, 2019-05-06:
Tried batch fire?

Tobbera, 2019-05-06:
I've tried all different settings. The only thing that did the trick was going two stroke.

Boogmeister, 2019-05-06:
Your heading is quite confusing too...

Are you after sequential injection or semi-sequential?

Tobbera, 2019-05-06:
I can update it.

I'm after sequential within the 360 degree without cam input. All inj and ign on individual channels.

Boogmeister, 2019-05-06:
What does that even mean though...

You can't have sequential without a phase input...

So what are you actually after.

Just explain it without calling it something.

Boogmeister, 2019-05-06:
Because it sounds like you want semi-sequential...

Tobbera, 2019-05-06:
I want the injector to fire in order. First number one, 120 degrees later number two, and another 120 degrees later number three.

I want the same with ignition.

I also want to adjust where in the 360 cycle the first injector fires. That made a very wierd behaviour that can be seen in the scopes in the forum thread.

Boogmeister, 2019-05-06:
Tried choosing 6 cylinder semi-sequential with wasted spark?

That should get you want you want.

Tobbera, 2019-05-06:
I can try that when I get home tonight.

Boogmeister, 2019-05-06:
That should do the trick.

Because you are running half a 6 cylinder engine ;)

Tobbera, 2019-05-06:
Fair enough. Will try.

ric355, 2019-05-06:
3 cylinder paired should also do what you want, as far as I can tell from reading the code.

LPG2CV, 2019-05-06:
What you are describing sounds like fully sequential, but because you don't don't have a cam  sensor, you can't have that with only a missing tooth crank wheel. This is because speedy can't tell if the the crank is on its power, or exhaust stroke.

Boogmeister, 2019-05-06:
> What you are describing sounds like fully sequential, but because you don't don't have a cam sensor, you can't have that with only a missing tooth crank wheel. This is because speedy can't tell if the the crank is on its power, or exhaust stroke.

No he just wants semi sequential.

He wants injectors to fire every 360 degrees but each injector 120 degrees apart.

It doesn't work though.

LPG2CV, 2019-05-06:
What you are describing sounds like fully sequential, but because you don't don't have a cam  sensor, you can't have that with only a missing tooth crank wheel. This is because speedy can't tell if the the crank is on its power, or exhaust stroke.

As well as the 2 stroke option, consider exploring odd fire.

Boogmeister, 2019-05-06:
> What you are describing sounds like fully sequential, but because you don't don't have a cam sensor, you can't have that with only a missing tooth crank wheel. This is because speedy can't tell if [...]

even fire engine though.... 120 degrees between each event.

LPG2CV, 2019-05-06:
Using odd fire, evenly spread at 120 degrees may run a different part of the code. Worth exploring

LPG2CV, 2019-05-06:
Just loomed at the code, and odd fire appears to only affect spark and not injection. This only looking at the setup and not looked at when its running.

ric355, 2019-05-06:
> Just loomed at the code, and odd fire appears to only affect spark and not injection. This only looking at the setup and not looked at when its running.

Yes this is correct odd fire is about ignition not injection. I've been meaning to mention this as I've seen a few people refer to odd fire in this context and its not quite correct.  You *can* alter the injector close angle for each channel though - it's in the injector characteristics dialog.

ric355, 2019-05-06:
btw my earlier suggestion of 3cyl paired is no good due to the TS limitations but I'm pretty sure 6 cyl paired as suggested by someone else will be suitable.

LPG2CV, 2019-05-06:
Also, keeping an eye on required fuel and VE table.

Tobbera, 2019-05-06:
I just tested 6 cyl, 4 stroke works. See the thread [https://speeduino.com/forum/viewtopic.php?f=19&t=2717&p=34439#p34439](url)

Tobbera, 2019-05-06:
But it still boils down to the 3 cylinder code being not right. It should be possible to choose 3 cylinder, 4 stroke, and get the desirable behavior, right?

Boogmeister, 2019-05-06:
It may be a limitation of TunerStudio again...

It might be something that MS can't do so was never setup in Speeduino correctly as you can't select it as an option?

Dunno.

Glad that 6 cylinder worked. I expected it would lol.

Tobbera, 2019-05-06:
Yes, thank you!

But think this needs to be looked into considering the weird behviour I got when running 3 cyl and tried to move the fire angle:   https://speeduino.com/forum/viewtopic.php?p=34391#p34391

Boogmeister, 2019-05-06:
I think i would prefer if that option was just locked out so it can't be used and instead be advised to use 6 cylinder option instead.

It works. No need to add or make any changes to code.

Tobbera, 2019-05-06:
But shouldn't it be possible to choose the 3 cylinder option and have it working in semi-sequential? And as you can see in my scopes in the above link, trying to change the fire angle there is really strange things happening with the injections.

Boogmeister, 2019-05-06:
Like i said earlier...

If it's something that MS can't do then TS won't let it happen.

Nothing we can do about that as it's a limitation of TunerStudio.

Tobbera, 2019-05-06:
But I don't think the limitation is within TS (and if it is, it can be worked around by ignoring the warning and burning anyway).

The problem is that we cant use the three cylinder semi-seq option with Speeduino firmware. And if trying to change the fire angle things really get out if hand.

ric355, 2019-05-06:
> But shouldn't it be possible to choose the 3 cylinder option and have it working in semi-sequential? And as you can see in my scopes in the above link, trying to change the fire angle there is really strange things happening with the injections.

You may have misunderstood what semi-sequential is in Speeduino's context. If you read the wiki it says that semi-sequential is like paired except the channels are mirrored - so for paired on a 4 cyl it uses 2  channels and each channel fires a pair of cylinders. For semi seq on a 4 cyl it fires all four channels but 1&4 are fired at the same time and 2&3 are fired at the same time.

Now map that to a 3 cylinder motor in semi-sequential mode with 3 injectors and if you monitor the first three injection channels you'll see that channel 1 fires at the same time as channel 4, but you're not interested in 4 so you don't see it. Then channel 2 fires at the same time as channel 3 and you see both of these. Then assuming you've said there are three injectors channel 3 will also fire independently at the "correct" angle. This is why you see these weird looking double injections in one of your images.

There are limitations on what can be configured in TS because some configurations just don't make sense.

LPG2CV, 2019-05-06:
Just read that, and my brain hurts. :o)

Tobbera, 2019-05-06:
Thank you for that @ric355 . 

Maybe the different options needs to be documented more extensively to explain what they actually do.

However, looking at one of the scopes, there is even three squirts on the third channel. https://speeduino.com/forum/viewtopic.php?p=34391#p34391
This is all depending och moving the closing angle around. Sometimes its one, two or three. This leads me to say there is a bug in the code.

And, isnt it desirable to be able to run the injectors "in sequence"? That is 1st then 2nd then 3rd injector divided over the 360 crank revolution. Since all works fine when wither going two stroke or 4 stroke 6 cylinder, I want to believe that the code is not correct when choosing 3 cylinder 4 stroke.

ric355, 2019-05-07:
> Thank you for that @ric355 .
>
> Maybe the different options needs to be documented more extensively to explain what they actually do.
>
> However, looking at one of the scopes, there is even three [...]

No objection to there being better documentation but you have to remember that what you've done here in the first place is to burn in a configuration that TS told you was invalid. In that circumstance you don't even know whether all of the settings you gave it were actually sent to the ECU, so it could be running some weird combination.

I can't account for the three injections without seeing the settings you used. It isn't immediately obvious where the third one would have come from on what I presume is channel 3 but I suspect it might be related to an invalid combination produced by TS not sending all of the values it says are in error.

I'm fairly sure that the code is set up right to do 3 cylinder 2 squirts per cycle sequential which I think is what you're asking for; it's just that TS does not allow it to be configured. I might have a look later and see if it works if configured outside of TS.

Tobbera, 2019-05-07:
Thank you. Is it beneficial to run two squirts per 360 degrees rather than one?

Boogmeister, 2019-05-07:
2 per cycle is once per 360 degrees...

A cycle is the full 720 degrees involve in a 4 stroke engine.

Your whole issue may come down to TS says NO! 

Which we can't do anything about and it's unlikely anything will change.

ric355, 2019-05-07:
> Thank you. Is it beneficial to run two squirts per 360 degrees rather than one?

Cycle = 720 degrees on a four stroke, but in fact full sequential always puts all of the fuel in with 1 squirt. so in fact it's 1 squirt per 720 degrees not 2.

noisymime, 2019-05-07:
Just to add a glimmer of hope here, I've been speaking with Phil from EFI Analytics and there is a chance that he would be willing to change TunerStudio to disable these checks on the dialog. This would free things up to support any combination (That Speeduino supports) without the current errors being given. 

There's not much I can do about this in the short term, but hopefully it's coming.

ric355, 2019-05-07:
> Just to add a glimmer of hope here, I've been speaking with Phil from EFI Analytics and there is a chance that he would be willing to change TunerStudio to disable these checks on the dialog. This [...]

There is something you could do, but it is a little bit hacky. You could add some extra speeduino specific settings to the dialog that allow the base options to be overridden. You could make these as specific as you like (e.g. just for 3 cylinder support). Then these extra settings would override the base ones during initialisation. Not pretty but it would work.

As suggested earlier, I hard coded the following settings to remove any inconsistency in burning after an error message;
    //RM mods 
    configPage2.injLayout = INJ_SEQUENTIAL;
    configPage2.nCylinders = 3;
    configPage4.sparkMode = IGN_MODE_SEQUENTIAL;
    configPage2.nInjectors = 3;
    // end RM mods 

And it works exactly as you would expect it to - 3 injection pulses over the 720 degree cycle on each of channels 1, 2, and 3.  I can actually get it to do this by burning after the error message as well, incidentally but I imagine that approach to be an unreliable method.

For the time being the settings can be hard coded in the firmware if the OP wants to make progress. I note that req_fuel is doubled when this config is run, so that's a potential gotcha for tuning depending on how the TS dialog behaves post error message.

Tobbera, 2019-05-08:
> Just to add a glimmer of hope here, I've been speaking with Phil from EFI Analytics and there is a chance that he would be willing to change TunerStudio to disable these checks on the dialog. This [...]

I got about the same reply from them:

> Hi,
>                 Yes that widget was build for MegaSquirt and it seems Speeduino now supports some different combinations. We are working with Josh to come up with a solution.
> 
> 
> Thanks,
>     EFI Analytics Software support.

noisymime, 2019-06-24:
This is now ready to go in the latest firmware. It allows any combination of squirts/cylinders, but you'll also need to grab the latest TS beta for it to run.

noisymime, 2024-02-05:
Closing as this has been implemented and tested.

closed 2024-02-05 by noisymime