speeduino/speeduino #228: TunerStudio not recognising pages 10 and 11

TradieUte, 2019-05-15:
The code at line 183 of comms.ino only works with pages 0 - 9.
To allow TS to work with pages 10 - 15,  the following code (or similar) is required.

//        if (currentPage >= '0') { currentPage -= '0'; } // single byte page id - only works 0 - 9
        if ((currentPage >= '0') && (currentPage <= '9')) // 0 - 9
        {
          currentPage -= 48;
        }
        else  // 10 - 15
        {
          if ((currentPage >= 'a') && (currentPage <= 'f'))
          {
            currentPage -= 87;
          }
          else
          {
            if ((currentPage >= 'A') && (currentPage <= 'F'))
            {
              currentPage -= 55;
            }
          } 
        }

noisymime, 2019-05-16:
Are you actually seeing this cause a problem? 

Whilst you're correct that these changes would be needed for setting the page in that function, this function itself (IE Set page via the 'P' command) shouldn't be being used any more. Sometime back this was switched over to the new `r/` and `w/` command modes, which don't use the 'P' command at all. 

Just a guess, but is this perhaps on something other than TunerStudio?

TradieUte, 2019-05-16:
Trying to read the tables with gtkTerm, found that couldn't read above table 9 with P command. The current code only supports t0 to t9.Can the new commands be used to display tables using gtkTerm?Was actually using TS mini terminal, when issue first noticed. Assumed (I now think incorrectly, from your reply) that the issue would also affect TS operation.Sent from Samsung tablet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/noisymime/speeduino/issues/228?email_source=notifications\u0026email_token=AJ5OAQE3Q5R5SYVTYA5OJMLPVSWVHA5CNFSM4HNAERLKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODVQLE4I#issuecomment-492876401",
"url": "https://github.com/noisymime/speeduino/issues/228?email_source=notifications\u0026email_token=AJ5OAQE3Q5R5SYVTYA5OJMLPVSWVHA5CNFSM4HNAERLKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODVQLE4I#issuecomment-492876401",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

noisymime, 2019-05-16:
Yes, you can definitely still read them with something like gtkTerm, but you need to use the new commands. 

To read values from a page now requires you to send a `p` command (Note that this is LOWER case) followed by 6 bytes:
* First byte is ignored, so just send a 0
* Second byte is the page number you want to read from
* Bytes 3 and 4 are the offset in the table you want to start reading from. These are combined together to give a single 16-bit int value in order to allow for offsets above 255
* Bytes 5 and 6 are the length of data you want to receive. These are combined same as bytes 3 and 4

I'm going to add your suggested changes to the old P (upper case!) command though, which will allow legacy systems to still read those extra pages.

closed 2019-05-16 by noisymime

TradieUte, 2019-05-16:
Many thanks for the info Josh.Tried the 'p' command earlier, without any luck. Can't find a way using gtkTerm to send non-ascii values. Using ascii chars, gives a page not implemented error.The protocol appears to require 0x0 for zero, whereas I am sending 0x30. Can you provide an ascii command that reads page 2, offset 0, length 256, please. (If it's possible, I have to be doing something wrong, but I just can't pick it.) Or do i need to find another tty?Sent from Samsung tablet.

noisymime, 2019-05-16:
Hmmm that's a good question. If you're only able to send ascii data, then you might be out of luck using the new method.

That said, I've applied the changes to the old `P` command now, so that should be working again in master.

TradieUte, 2019-05-16:
Thanks Josh, confirming what i thought. The P comand works now does what I need.Sent from Samsung tablet.

commit 887817f8 (2019-05-16): Add support for pages 10+ using the legacy `P` command. Fixes #228