speeduino/speeduino #240: Cranking Bypass does not fall back to cranking mode

turboedge, 2019-08-05:
When using cranking bypass the output correctly transitions from cranking mode to running. However if the engine stalls the output should immediately go back to the cranking mode like it is when you first turn the key to the on position.  Unfortunately this seems to only happen less than half the time, the rest of the time the output stays in the running mode.

My engine will restart because Speeduino works to control the timing itself, but in my case I am using the output for two purposes. The other to control the wide-band heater. So I would like it if it would turn off so that the wide-band shuts down when the engine is not running as intended.

Thanks!!

brunob45, 2019-08-05:
With the current code, the bypass will only reset when the speeduino reads an RPM between 0 and cranking RPM limit, which seems to happens not so often.
A quick fix would be to add the line 
`if(configPage4.ignBypassEnabled > 0) { digitalWrite(pinIgnBypass, LOW); }`
under line 134 of speeduino.ino.
This will make the bypass output go LOW while the engine is stopped.

noisymime, 2019-08-05:
Yeah that was pretty much exactly what I was thinking, should be very straightforward. I'll test it tonight and get it added.

noisymime, 2019-08-05:
Hmmm, so I went to add this and then realised that the existing code actually works the opposite way around. 

`if(currentStatus.RPM > currentStatus.crankRPM) //Crank RPM in the config is stored as a x10. currentStatus.crankRPM is set in timers.ino and represents the true value
        {
          BIT_SET(currentStatus.engine, BIT_ENGINE_RUN); //Sets the engine running bit
          //Only need to do anything if we're transitioning from cranking to running
          if( BIT_CHECK(currentStatus.engine, BIT_ENGINE_CRANK) )
          {
            BIT_CLEAR(currentStatus.engine, BIT_ENGINE_CRANK);
            if(configPage4.ignBypassEnabled > 0) { digitalWrite(pinIgnBypass, HIGH); }
          }
        }
        else
        {  
          //Sets the engine cranking bit, clears the engine running bit
          BIT_SET(currentStatus.engine, BIT_ENGINE_CRANK);
          BIT_CLEAR(currentStatus.engine, BIT_ENGINE_RUN);
          currentStatus.runSecs = 0; //We're cranking (hopefully), so reset the engine run time to prompt ASE.
          if(configPage4.ignBypassEnabled > 0) { digitalWrite(pinIgnBypass, LOW); }
        }`

When the engine is cranking, the output goes LOW, and when it's running it goes HIGH. This seems backwards to me, but I can't recall now whether it was done for a specific reason. 

There's actually a setting in the .ini file called `ignBypassHiLo` that looks like it was going to be for setting this direction, but it was never actually used. 

@turboedge Does that match the behaviour you're seeing at the moment?

turboedge, 2019-08-06:
Yes, when key on engine off it starts out low and goes high when the engine starts. The GM late model HEI apparently works at logic level this way. You are right, it seems a bit backwards, but is how it works. I have re-purposed the second ignition channel driver for this instead of using the arduino output directly.

When the engine falls out of the running state, without turning the key off, it only rarely brings the output back to low.

brunob45, 2019-08-06:
This would be a nice opportunity to use a "state machine". It is an easy way to track actions to do on state change (cranking->running, running->stalled).

commit 732718f2 (2019-08-06): Set ignitionBypass pin back to low when engine off. Fixes #240
closed 2019-08-06 by noisymime