speeduino/speeduino #245: triggerFilterTime = 0 in basic distributor decoder

thomas41546, 2019-08-16:
https://github.com/noisymime/speeduino/blob/3d3437406dc4694ad29e51a00dc0a136b67da5df/speeduino/decoders.ino#L644,L646
`  triggerFilterTime = 60000000L / MAX_RPM / configPage2.nCylinders; // Minimum time required between teeth
  triggerFilterTime = triggerFilterTime / 2; //Safety margin
  triggerFilterTime = 0;`

Looks like the triggerFilterTime is incorrectly set to 0 in the triggerSetup_BasicDistributor function.

noisymime, 2019-09-05:
This is only set to 0 on startup, which ensures that the first pulse is always received and not filtered out because there's not a last value. The filter time is then set and updated when `setFilter(curGap)` is called within the triggerPri ISR.

noisymime, 2019-10-14:
Closing as this is working as intended

closed 2019-10-14 by noisymime