speeduino/speeduino #254: Flexfuel timing issue and small improvement suggestion

muuris, 2019-09-07:
I noticed an issue with unstable ignition advance when using negative flexfuel ignition advance values on 201905 firmware and looking at the code, the same issue remains in 201909. The calculation of ignFlexValue is done as unsigned byte so it overflows (the same issue as earlier with ignCLTValue, and fix would also be similar). 

For the use case of this, I'm going to use ethanol normally and only use pump gas blends when E85 is unavailable, so it would be more user-friendly to look  ignition maps as they are and knowing that timing will be retarded when using blends -- instead of having ignition map for gas and remembering that these readings are advanced further for my normal tune.

Also, if idle is tuned with very moderate advance, the fixed ethanol % / advance in degrees throughout rev range can be problematic, so some additional parameters for this could be handy.

noisymime, 2019-09-18:
Can confirm this. 

That timing adjustment wasn't originally intended to have negative advance values for flex, but these values were allowed in TunerStudio. 

I will incorporate the fix for this into the larger piece to change the flex ignition advance curve into a 6x6 3D map. This will not only allow for negative values, but will mean that the advance amount can be controlled not only based on eth% but also RPM.

muuris, 2020-04-15:
I've made a PR to fix the original issue with current flex ignition functionality.

About larger flex change, I think load is just as important as RPM. In my opinion the proper way of doing this would be using current 2D tables for fuel, ignition and boost as blend tables (to get interpolation value) between two 3D tables. There is already second full resolution fuel table, but on Mega2650 there aren't really resources to make another full ignition and boost tables. However, at a glance there seems to be enough free EEPROM after config11 to have second 6x6 maps for fuel, ignition and boost. This wouldn't be perfect, but would give much more control than the current functionality, and would be much more than many lower end ECUs offer. I believe many of us have a "main" fuel and are not constantly driving on blends, this would do just fine, as users can define bins so they have resolution where it's needed the most.