speeduino/speeduino #267: DFCO malfunction if staged injection is enabled

Thommmm, 2019-10-18:
If staged injection is enabled DFCO malfunctions. During decel both duty cycles go to maximum due to a bug in the staged injection calculation.

"Problem" in the code:
`currentStatus.PW1 -= inj_opentime_uS; //Subtract the opening time from PW1 as it needs to be multiplied out again by the pri/sec req_fuel values below.`
If currentStatus.PW1 is 0 (because DFCO makes it 0), the value rolls back because a value bigger then 0 is subtracted (inj_opentime_uS)
My temporary fix is to include a check in the staged injection calculation to see if PW1 is bigger then inj_opentime_uS:
`if( (configPage10.stagingEnabled == true) && (configPage2.nCylinders <= INJ_CHANNELS) &&  (currentStatus.PW1 > inj_opentime_uS) )`
Haven't tried it but I think it should work

[DFCO enabled - staged injection.txt](https://github.com/noisymime/speeduino/files/3743266/DFCO.enabled.-.staged.injection.txt)

noisymime, 2019-10-18:
Thanks for the report. I'll get this fixed shortly

brunob45, 2019-10-18:
I looked into that part of the code, and I think the injector open time shouldn't be included into the stagingSplit calculation 
```
uint32_t tempPW1 = (((unsigned long)currentStatus.PW1 * staged_req_fuel_mult_pri) / 100) + inj_opentime_uS;
...
currentStatus.PW1 = ((100 - stagingSplit) * tempPW1) / 100;
```
should be 
```
uint32_t tempPW1 = (((unsigned long)currentStatus.PW1 * staged_req_fuel_mult_pri) / 100);
...
currentStatus.PW1 = ((100 - stagingSplit) * tempPW1) / 100 + inj_opentime_uS;
```
The way the code is will make the open time smaller with greater split, but the open time is constant

closed 2019-10-22 by noisymime

noisymime, 2019-10-22:
Both issues mentioned here should now be resolved. If you're still seeing issues please comment back and I'll reopen

commit 9b967070 (2019-10-22): Fix for potential underflow in staging code when DFCO enabled. Fixes #267