speeduino/speeduino #270: FlexBoost Table axis is not assigned properly

brunob45, 2019-10-28:
In init.ino, lines 122-125, the values of `flexBoostTable` are assigned to `values16`, but the bins to `axisX`
```
flexBoostTable.valueSize = SIZE_INT;
flexBoostTable.xSize = 6;
flexBoostTable.values16 = configPage10.flexBoostAdj;
flexBoostTable.axisX = configPage10.flexBoostBins;
```
This make the function `table2D_getValue` to return random values as it uses a table that was never initialized (`flexBoostTable.axisX16`)

Proposed fix : replace `axisX16` by `axisX` everywhere because the former is never used anywhere.

noisymime, 2019-10-29:
I've done nearly a full rewrite of the 2D table lookup to address this and pull out all the crappy checks for 8-bit or 16-bit values. Unfortunately there was really no easy way to do this with the existing 2D table lookup method.

> replace axisX16 by axisX everywhere

You can't do this if you ever want to have an axis with 16-bit values. 

> Or add a second field axisSize to specify the correct size.

This works, but takes up further memory and still leaves all the messiness. 

The new method takes considerably less memory, but does have some CPU overhead to cache the arrays. It should be considered SUPER experimental at this point until I've tested all instances. Do feel free to start testing though.

noisymime, 2019-11-07:
This was fixed in 91206ac90e5cb25a27397bd7f3426bf6863a0599

closed 2019-11-07 by noisymime