speeduino/speeduino #277: Add ability to override BARO readings

XPbIM3, 2019-11-07:
BMW L-jetronic relay puts power on ECU only when cranking.  This often ruins BARO readings. Standart pressure menus does not offer a chance to override BARO readings with constant.

XPbIM3, 2019-11-13:
A workaround is to set any free Analog input to use as external BARO pin, and assign a flat calibration 0v-5v as a constant 100kpa. (or 95kpa, or whatever in Your case)

muuris, 2019-11-14:
How about just taking Speeduino power supply from ACC (key position II) distribution? Better to fix the actual cause of the problem than leaving the problem and finding workarounds (and disabling baro corr).

noisymime, 2020-01-22:
What values for Baro do you see when cranking? 

There's already a sanity check for baro readings that are taken from the MAP sensor on startup. Only values between 87kPa and 108kPa (Inclusive) are permitted. If the value at startup falls outside that range then it will first attempt to use the last valid baro reading (Stored in EEPROM) and if that turns out to be invalid as well (Based on the same range) then it will default to simply using 100kPa.

That seems like it would already cover your issue?

XPbIM3, 2020-07-31:
>That seems like it would already cover your issue?

Mostly.
Normal pressure in my location is 96-97kpa.
Sometimes a false reading of ~90kpa or lower is achieved. i assume this happens when:

-Attaching a PC to USB port of arduino,  so engine will stall while TunerStudio is syncing. (by the way,  is there a chance to sync TunerStudio on-the-fly without engine stall? ) 
-during the large RPM drop when engine is almost died or died, and then re-started shortly(on Mercedes you have to switch off the ignition to be able to start car again,  so ECU re-power is performed).  
-Maybe on a tow-start,   but that seems unlikely possible to Me... Only if try to power-reset an ECU with ignition key during towage.