speeduino/speeduino #281: Add MAF support

XPbIM3, 2019-11-13:
That would be a please to have a chance to use a Mass Airflow Sensor as a Load sensor.
There is a need to have a calibration lookup curve for voltage against air mass. And then use a load_factor vs RPM table (similar to VE table) for fuelling.

pazi88, 2019-11-13:
Using airmass as load directly isn't very great idea because the changes in airmass are huge between different driving conditions. Like turbo engine can easily suck 2000kg/h of air when running at full blast. and then suck maybe 100kg/h of air at WOT close to idle RPM. Stock ecus deal with this calculating airmass per stroke and using that as load rather than using measured airmass directly as load. The calculation itself isn't that hard. For example Siemens ms41 uses this load calculation: "Load (mg/stroke) = Airflow (kg/hr) / Engine Speed (RPM) * 5555"

So basically adding MAF support would need that mentioned addition of voltage vs airmass curve. And then load calculation from airmass and RPM.

XPbIM3, 2019-11-13:
Agree.
Nissan ECUs usess a TP_scale  where air mass and RPM are involved.

XPbIM3, 2019-11-13:
> TP = Airflow lookup (VQ map) / RPM * K constant

TBAMax, 2020-07-30:
[image]
It may be quite viable to just 3D print the MAF sensor housing and install MAF sensor.

DeeEmm, 2020-07-30:
I've been playing with MAF sensors as part of my flow-bench project. ( DeeEmm/DIY-Flow-Bench ) There's a few learnings there that might be of use.

For full compatibility you need to consider that there are different types of sensors. Some operate on voltage, some on frequency. So there's signal conditioning and code differences depending on which sensor you are dealing with.

Often the (voltage) output characteristic is linear, which lends itself well to using a mathematical solution for converting MAF output into something useable, however in our testing we have found that not all sensors are in fact linear, so lookup tables then become the only (sensible) option.

There's some sensor data available online, but there's not much. So to be able to provide support for a specific sensor, ideally you need to know the OEM lookup data. In this respect there are going to be a bunch of sensors that you will not be able to support simply as data is not available. 

From a DIY perspective you can simply select a suitable sensor for the application, but when trying to retrofit to an existing MAF the lookup data might not be available.

When 3D printing your own housing you have to take the diameter of the housing into account as it affects the flow rate. Unless the housing is the same diameter as the OEM application, the published figures will be inaccurate, so you will need to transpose them. There's plenty of sensors available with housings for DIY use.

My advice when choosing a MAF is find one that has published lookup data or output graphs.

/DM

XPbIM3, 2020-07-31:
The Frequency type sensors are early toyota vortex sensors.
Present days sensors are voltage, as i know.

dascons, 2020-07-31:
Frequency and voltage are both used in all generations. There are a huge
number of cars (and air flow meters) out there and both types are still in
production.
It is not going to be easy to get a good calibration for many sensors
(especially if people are going to be swapping the diameter housing around
as stated before) and there are plenty of AFM's that vary by like one
number on the P/N but have reasonably different calibration.
It will probably have to be similar to the MAP sensor calibration drop down
where there are a few known sensors (with the right diameter pipe also) and
a custom mode where best case will be extracting data from a factory ECU
(hptuners etc)

noisymime, 2024-03-14:
Currently no intention to implement this, either in firmware or hardware in any official way. 

Closing unless anyone wants to volunteer to take this on.

closed 2024-03-14 by noisymime